motorapp/Galil #13: IOC zeroes encoder on startup

djvine, 2020-05-11:
Hi,

I have a question about how the Galil support handles encoder value on IOC startup/restart. 

I have an incremental rotary encoder on one of my linear axes. If I don't ask autosave to restore it what should the default behaviour of the motor record be:
1) read the current value of the encoder from the Galil and use that?
2) zero the encoder

We recently moved everything to a docker container, upgraded to the latest synapps and moving from galil 3-5 to 3-6 i now get (2) where I used to get (1). But I don't know what I did to cause that. For my application (1) would be preferred. I could use autosave but I didn't need to do that in the past. I don't think the docker is responsible.

My configuration is:
galil 4183 
base 3.15.6
synapps - latest tagged version from assemble_synapps.sh

FreddieAkeroyd, 2020-05-11:
Hi, I think the issue may be in the epics motor record included with the newer synapps package you are using. We came across a similar issue when we updated to motor 6-11 which is described in https://github.com/epics-modules/motor/issues/85 - the autosaved position is always restored, but as we didn't autosave positions the encoder was set to zero. You can patch the motor code with something like https://github.com/ISISComputingGroup/EPICS-motor/commit/bdcdaad8f7e687cce4d0f3f28a8261c9f10fcfb9 to restore the old behaviour, there is a ticket in progress to make this behaviour controlled by a new motor record field https://github.com/epics-modules/motor/issues/151 but I don't think this change is released yet

tboegi, 2020-05-12:
Out of interest:
May I ask what values do you use for the RTRY and RMOD field in you applications ?
And why ?

motorapp, 2020-05-12:
Hi,
The design behaviour is (1)
I am unable to reproduce (2) on CentOS 6.10 64 bit with:
base-3-15-6
asyn-4-37
motor-7-1
autosave-5-10
Galil-3-6-54 (check $(P)DRIVER_MON or GalilController.cpp line ~406 driverVersion)

I've not received similar bug reports on the Galil driver previously.

I use RTRY on stepper motors only with RMOD set to unity mode.

FreddieAkeroyd, 2020-05-12:
For most of our axes we generally have RTRY == 10 and RMOD == 0 (Unity). 

The encoder zeroing issue I saw with motor 6-11 only happens when  RTRY > 0,  RMOD != In-Position,  (UEIP || URIP) != 0

tboegi, 2020-05-12:
I may have time to look into 
https://github.com/epics-modules/motor/issues/151
and implement a path the next days (no 100% promises)

motorapp, 2020-05-12:
Thank you, I can now reproduce the error.  I'll investigate further.

djvine, 2020-05-12:
Thanks @FreddieAkeroy. That's a much better description than I was able to provide.

motorapp, 2020-05-14:
Investigating under conditions that create this behaviour, I've found:
1.  The GalilAxis setPosition is being called at ioc start with value 0.  In this case the Galil driver is correctly setting the encoder register to zero as requested by upper layer motor, asyn layers
2.  The value 0 being pushed to the encoder register comes from the motor record DVAL set at record definition time (in the absence of autosave restore).  Change DVAL at record definition and the restore value will change to match
3.  This behaviour is caused by issues in devAsynMotor.c init_controller method and was introduced by a change on July 30 2015 titled "Bug fix for target position (VAL/DVAL/RVAL) initialization error when the motor record is configured to do retries."
Introduction of use_rel in the if statement allows 0 to be restored
See also
https://github.com/epics-modules/motor/blob/master/docs/RELEASE.txt line 196
Additional notes
pmr-dval inside init_controller is the motor record dval field, and the value is not from the controller
init_controller method is also missing GET_INFO after the LOAD_POS.  GET_INFO is required after LOAD_POS so that rbv is not stale, contains the correct restored value, when val is set to rbv at motor record initialization

Thoughts?

FreddieAkeroyd, 2020-05-14:
I am not sure I quite follow the logic discussed at line 196 mentioned above. If retries are needed, or if motors stall even without retries, then the absolute motor position and the encoder position will start to  differ. But is this a big problem - you know where it really is from the encoder. The only issue we have had in this situation with a galil is that the controller soft limits are based on motor not encoder position, so if the motor and encoder values get out of step though the position is still reported accurately the controller may signal a soft limit violation earlier than expected (which we now have a check for and can resyn motor position). So was the original problem there that on boot the motor rather than encoder position was read from the controller and used?

FreddieAkeroyd, 2020-05-15:
So I guess the issue is that if VAL was read from controller on startup it may look a bit strange in the GUI display if there had been a lot of retries, however everything would work OK as when a new VAL was set it would issue a relative motor move based on the new VAL and the current RBV. If you stop a move that is in progress, VAL is set equal to RBV, wouldn't setting VAL = RBV on boot be the thing to do in the motor record (though it looks like it is missing a GET_INFO as you mentioned above)? The problem we had was that applying an autosaved VAL seemed to redefine the motor and encoder positions, and as we dodn't autosave VAL this resulted in them being zerod.

tboegi, 2020-05-15:
Disclaimer: I don't have a Galil controller - but recognize the problem(s)

Actually the startup looks like this:
- (1) The st.cmd creates a controller, then all axes.
- (2) The poller() reads the current information from the controller:
   position, encoderPosition and status.
- (3) The motorRecord is initialized.
   in devMotorAsyn/init_controller() the device support reads the motor position:
    double position = pPvt->status.position;
- (4) (Simplified , see below)
     If the position is 0 +/- RDBD, we keep it and it is put into the RBV and later into
        .VAL
      If the position is 0, we assume that the controller is rebooted (and the IOC)
      and fall back to the auto-saved  value from .DVAL
      That position will be loaded into the controller, using LOAD_POS.
      Later the poller will update RBV, so that RBV and VAL match.
    And now the un-simplification:
    The assumption "if the controller has a position, we use it" was changed in  
   https://github.com/epics-modules/motor/commit/3090983c313c57efdcd7c788ba234e775d300e06
   That commit seems to assume, that people always use autosave, to my understanding.

I can see different and independent problems here:
- (A) We found the timing between (2) and (3) extremely fragile.
  Sometimes the poller() has managed to poll the first axis, but not the second or third one.
  As a result, I changed the logic in the motorRecord:
  Keep the record in UDF state, and sync .VAL from .RBV when the controller comes online.
  Another attempt would be that the createAxis() polls it's axis, before it returns.

- (B) We can't choose which value we want to have when the motorRecord is
  initialized. Either:
   - (a) The .DVAL (from autosave)
   - (b) the motorPosition or 
   - (c) the encoderPosition
  A suggestion https://github.com/epics-modules/motor/issues/151
  Reading all this, it feels as if the new .RSTM field should have
   useVAL
   useMotorPosition
   useEncoderPosition
   useZero
   (But we are getting slightly off-topic)

- (C) When the Galil uses the motorPosition for internal softlimits,
   and not the encoderPosition, then I have 2 solutions in mind:
    - fix it in the Galil SW
    - Synchronize the motorPosition from the encoderPosition from time
       to time. Probably when the motor is standing still.
       Before or after a move.
       There has been a fix being done for one of the motor drivers, but I don't remember
        which one

FreddieAkeroyd, 2020-05-15:
Hi @tboegi thank you for the summary of how the position restore works. The commit you reference certainly does make using autosave a necessity, but we don't use autosave on positions as we can't assume they are still correct after a power outage and so don't want to automatically try and fix things. 

With regard to (C) i don't think there is a way to get the galil controller to use encoder rather than motor position for applying its controller based limits. We took the approach you mention of resyncing motor to encoder at start of a move, but we don't do this for all axes as there were some concerns raised e.g. if the encoder failed you may then end up losing a good motor position that you could have then used by switching to open loop.

motorapp, 2020-05-16:
Hi,
Thank you @thoegi, for taking the time to write the restore description.
Around points 3) and 4) the description does not include the use of the use_rel variable in the below if statement (devAsynMotor.c init_controller function)

    if ((use_rel != 0) ||
        ((fabs(pmr->dval) > rdbd) && (pmr->mres != 0) && (fabs(position * pmr->mres) < rdbd))
       )

And the fact that it's use_rel OR in the above if statement.

The conditions that cause the reported issue are:
1. RTRY !=0, and
2. RMOD != motorRMOD_I, and
3. ueip or urip set true

These conditions cause use_rel to be set 1 by this earlier statement
    int use_rel = (pmr->rtry != 0 && pmr->rmod != motorRMOD_I && (pmr->ueip || pmr->urip)); 

The conditions mentioned in the restore description points 3) and 4) are being circumvented since use_rel has the value of 1.
Otherwise the original logic, that matches the description seems good.  I understand and appreciate this change was made originally to address short-commings in the Galil EPICS driver.  I apologize for the inconvenience caused at that time.

Keeping the step count register, and encoder register consistent is a job of the device support.  Not all controllers function with a dedicated step count, and encoder count register in the way a Galil operates.  For example on a Powerbrick, or Geobrick we typically use an Encoder Conversion Table Entry along with a c structure or IVar.  The values inside these registers can be computed micro-step position, BISS position, or incremental position.  For this reason, I consider the step count, and encoder register synchronization an implementation, and manufacturer specific detail.

A feature to synchronize the step count register with the encoder register has been put into the Galil EPICS driver, and it was completed on 31/08/2016.  The step count register is synchronized to the encoder at the end of each move, and the moving flag is held true during this process.  Later, whilst the moving flag is false, if the encoder value drifts more than rdbd, another synchronization will be done.  The synchronization method is for both incremental, and absolute encoders.  The feature is designed to stop the drift in "agreement" between step count, and encoder register.  The feature is specific for "encoded" steppers.  Galil also offer closed loop stepper firmware, and the operation becomes remarkably similar to a Geobrick at that point.

It appears epics-modules/motor@3090983 was put in place (July 2015) due to an issue with step count, and encoder register "agreement" drift encountered when using the Galil EPICS driver.  A valid issue, and a real concern.  Since that time, the issue has been addressed in the Galil driver.  Perhaps epics-modules/motor@3090983 should be reviewed now circumstances have changed?

The restore process in devAsynMotor.c init_controller has additional issues as mentioned.  These issues appear related to the historical asyn motor conversion from motordevCom.cc.  From motor init_record, device init_record is called that in turns calls devAsynMotor.c init_controller.  The devAsynMotor.c init_controller sets a new position on the hardware by LOAD_POS, and later, with no thread control code in between, in motor init_record the val field is set to rbv (rbv is stale).  I recommend that devAsynMotor.c init_controller be altered so that after the LOAD_POS it asks device support for an update via GET_INFO.  Additionally, the initEvent should be signaled from devAsynMotor.c asynCallback when the command is motorUpdateStatus.  In this way devAsynMotor.c init_controller can do a LOAD_POS, ask and then wait (thread control code) for an update via GET_INFO before motor init_record sets val to rbv.  The original logic in motordevCom.cc does include the requested GET_INFO after LOAD_POS in motor_init_record_com.  We lose stuff sometimes... oops.

motorapp, 2020-05-17:
Hi All.  Thank you @tboegi for the restore description.

At ANSTO Synchrotron I've used Geobrick, Powerbrick, PMAC2, and Galil.  Galil is the only controller I know that has a main, and auxiliary encoder register per axis.  For encoded steppers generally the step count is stored in the auxiliary register, whilst the encoder count is stored in the main register.  Since other controllers operate differently, keeping the step count register in agreement the encoder register seems a job for the device support.  In this case, it means it's something the Galil EPICS driver should take care of.

A feature to synchronize the step count register with the encoder register has been put into the Galil EPICS driver, and it was completed on 31/08/2016.  The step count register is synchronized to the encoder at the end of each move, and the moving flag is held true during this process.  Later, whilst the moving flag is false, if the encoder value drifts more than rdbd, another synchronization will be done.  The synchronization method is for both incremental, and absolute encoders.  The feature is designed to stop the drift in "agreement" between step count, and encoder register.  The feature is specific for "encoded" steppers.  Galil also offer closed loop stepper firmware, and the operation becomes remarkably similar to a Geobrick at that point.

It appears epics-modules/motor@3090983 (July 2015) was in response to an issue with step count, and encoder register "agreement" drift when using the Galil EPICS driver.  Since that time, the issue has been addressed in the Galil driver.  Perhaps epics-modules/motor@3090983 should be reviewed now circumstances have changed?  Additionally, epics-modules/motor@3090983 is causing unintended side-effects, as reported by others.  Recommend the changes from epics-modules/motor@3090983 be removed.  Is further discussion required to reach agreement on this point?

I'd like to separate the LOAD_POS/GET_INFO issue I raised to another thread.  Agree? It may be the same block of code as the issue reported here, but it's complicating this discussion.  Also, it's not a big problem, since the Galil EPICS driver has a work around to this specific issue.

For any folks with Galil poller issues, please create a separate issue, or email me with your problem description.  I'll provide the best support I can.

FreddieAkeroyd, 2020-05-18:
Hi, the effect of the changes introduced in epics-modules/motor@3090983 have been discussed in  epics-modules/motor#85 and epics-modules/motor#151 and the current suggestion is to create a new motor record field to make this behaviour conditional with the default being the original (old) behaviour. As this is a motor record issue, I agree any further discussion should be held on one of the aforementioned tickets rather than here. If the galil is the only type of controller requiring something along the lines of the original change, then there may now be a case for reverting this change and not introducing the new field, but I don't know if this option might be useful for another controller model. 

Yes, the LOAD_POS/GET_INFO should de discussed separatey, probably as an issue in the motor record repository as it affects other controller types.

The only item I'd like to discuss further, not necessarily in this issue, is the mechanism for syncing and reporting of motor/encoder position. We are still using an older version of the galil dirver, so implemented a similar sync approach as described above ourselves, but we don't currently do this sync automatically on all axes as people had concerns it could hide developing mechanical issues. Does the galil driver keep a record of the total drift it has had to correct over time? This could be useful to archive as if the driver suddenly has to start correcting more often or by bigger margins then this may indicate developing mechanical issues.

motorapp, 2020-05-19:
Hi,
@djvine it seems this report is related to a motor record issue, so I will proceed to close this issue.
@FreddieAkeroyd as mechanical issues get worse over time the motor won't reach position after set RTRY and the MISS field will be set.  Over time the number of retries required to avoid a MISS will increase due to wear.  The Galil driver also has encoder stall detection for both steppers and servos that includes a direction consistency check.  Also good to keep in mind with open-loop encoded steppers that the motor record is attempting to reach encoder position (ueip=1), whilst the controller is using step count to get there.
@tboegi and all.  Thank you.
Kind regards,
Mark

closed 2020-05-19 by motorapp

djvine, 2020-05-19:
Thanks all for the detailed and thoughtful analysis. The EPICS community never fails to amaze me!