motorapp/Galil #15: error when start IO

LeeYangLBLBCS, 2022-05-18:
```
base) bl531@BL531BCS-L:/opt/epics/modules/motorGalil/Galil-3-0/3-6/iocBoot/iocGalilTest$ ../../bin/linux-x86_64/GalilTest st.cmd
#!../../bin/linux-x86_64/GalilTest
< envPaths
epicsEnvSet("IOC","iocGalilTest")
epicsEnvSet("TOP","/opt/epics/modules/motorGalil/Galil-3-0/3-6")
epicsEnvSet("SUPPORT","/opt/epics/modules/synApps_6_1_epics7/support")
epicsEnvSet("EPICS_BASE","/opt/epics/base-7.0.4")
epicsEnvSet("AUTOSAVE","/opt/epics/modules/synApps_6_1_epics7/support/autosave-R5-10")
epicsEnvSet("SNCSEQ","/opt/epics/modules/synApps_6_1_epics7/support/seq-2-2-7")
epicsEnvSet("SSCAN","/opt/epics/modules/synApps_6_1_epics7/support/sscan-R2-11-3")
epicsEnvSet("CALC","/opt/epics/modules/synApps_6_1_epics7/support/calc-R3-7-4")
epicsEnvSet("ASYN","/opt/epics/modules/synApps_6_1_epics7/support/asyn-R4-38")
epicsEnvSet("BUSY","/opt/epics/modules/synApps_6_1_epics7/support/busy-R1-7-2")
epicsEnvSet("MOTOR","/opt/epics/modules/synApps_6_1_epics7/support/motor-R7-2-1")
epicsEnvSet("IPAC","/opt/epics/modules/synApps_6_1_epics7/support/ipac-2-15")
cd /opt/epics/modules/motorGalil/Galil-3-0/3-6
## Register all support components
dbLoadDatabase("dbd/GalilTest.dbd",0,0)
GalilTest_registerRecordDeviceDriver(pdbbase)
cd /opt/epics/modules/motorGalil/Galil-3-0/3-6/iocBoot/iocGalilTest
### Scan-support software
# crate-resident scan.  This executes 1D, 2D, 3D, and 4D scans, and caches
# 1D data, but it doesn't store anything to disk.  (See 'saveData' below for that.)
dbLoadRecords("/opt/epics/modules/synApps_6_1_epics7/support/sscan-R2-11-3/sscanApp/Db/standardScans.db","P=IOC01:,MAXPTS1=8000,MAXPTS2=1000,MAXPTS3=10,MAXPTS4=10,MAXPTSH=8000")
dbLoadRecords("/opt/epics/modules/synApps_6_1_epics7/support/sscan-R2-11-3/sscanApp/Db/saveData.db","P=IOC01:")
# Configure an example controller
< galil.cmd
## uncomment to see every command sent to galil
#epicsEnvSet("GALIL_DEBUG_FILE", "galil_debug.txt")
#Load motor records for real and coordinate system (CS) motors
#Motor record version 6-9 and below
#dbLoadTemplate("$(TOP)/GalilTestApp/Db/galil_motors-v6-9down.substitutions")
#Motor record version 6-10 and up
dbLoadTemplate("/opt/epics/modules/motorGalil/Galil-3-0/3-6/GalilTestApp/Db/galil_motors-v6-10up.substitutions")
macLib: macro GALIL is undefined (expanding string $(GALIL)/GalilSup/Db/galil_motor-v6-10up.template)
filename="../dbStatic/dbLexRoutines.c" line number=268
dbRead opening file (null)
dbLoadRecords: failed to load '$(GALIL)/GalilSup/Db/galil_motor-v6-10up.template'
macLib: macro GALIL is undefined (expanding string $(GALIL)/GalilSup/Db/galil_motor-v6-10up.template)
filename="../dbStatic/dbLexRoutines.c" line number=268
dbRead opening file (null)
dbLoadRecords: failed to load '$(GALIL)/GalilSup/Db/galil_motor-v6-10up.template'
macLib: macro GALIL is undefined (expanding string $(GALIL)/GalilSup/Db/galil_motor-v6-10up.template)
filename="../dbStatic/dbLexRoutines.c" line number=268
dbRead opening file (null)
dbLoadRecords: failed to load '$(GALIL)/GalilSup/Db/galil_motor-v6-10up.template'
macLib: macro GALIL is undefined (expanding string $(GALIL)/GalilSup/Db/galil_motor-v6-10up.template)
filename="../dbStatic/dbLexRoutines.c" line number=268
dbRead opening file (null)
dbLoadRecords: failed to load '$(GALIL)/GalilSup/Db/galil_motor-v6-10up.template'
macLib: macro GALIL is undefined (expanding string $(GALIL)/GalilSup/Db/galil_motor-v6-10up.template)
filename="../dbStatic/dbLexRoutines.c" line number=268
dbRead opening file (null)

motorapp, 2022-05-18:
Hello,
Is the GALIL macro defined in the same RELEASE file as the modules listed below?
epicsEnvSet("AUTOSAVE","/opt/epics/modules/synApps_6_1_epics7/support/autosave-R5-10")
epicsEnvSet("SNCSEQ","/opt/epics/modules/synApps_6_1_epics7/support/seq-2-2-7")
epicsEnvSet("SSCAN","/opt/epics/modules/synApps_6_1_epics7/support/sscan-R2-11-3")
epicsEnvSet("CALC","/opt/epics/modules/synApps_6_1_epics7/support/calc-R3-7-4")
epicsEnvSet("ASYN","/opt/epics/modules/synApps_6_1_epics7/support/asyn-R4-38")
epicsEnvSet("BUSY","/opt/epics/modules/synApps_6_1_epics7/support/busy-R1-7-2")
epicsEnvSet("MOTOR","/opt/epics/modules/synApps_6_1_epics7/support/motor-R7-2-1")
epicsEnvSet("IPAC","/opt/epics/modules/synApps_6_1_epics7/support/ipac-2-15")

Once the RELEASE file is altered, the Galil EPICS driver should be recompiled to produce a new envPaths file in the iocBoot directory.

Could you please attach the files listed below?  So I may investigate further.
$(TOP)/config/GALILRELEASE
$(TOP)/configure/RELEASE

Thanks,
Mark

motorapp, 2022-05-19:
Thanks for the RELEASE files, I cannot see any errors in configuration.  Unfortunately, I cannot reproduce the fault yet, I will keep trying.  This is the first time this fault has been reported.

The critical line in the build is:
perl -CSD /opt/epics/base-7.0.4/bin/linux-x86_64/convertRelease.pl -t /opt/epics/modules/motorGalil/Galil-3-0/3-6 envPaths
called from the directory path
/opt/epics/modules/motorGalil/Galil-3-0/3-6/iocBoot/iocGalilTest

LeeYangLBLBCS, 2022-05-19:
intresting to know. The perl script, convertRelease.pl, seems to change a lot between different epics bases.
I don't know perl well to debug the scripts, but I compare the same scripts from base-3.15.6 and base-7.0.4, and found they are quite different:
[perl_scripts.zip](https://github.com/motorapp/Galil-3-0/files/8733830/perl_scripts.zip)

motorapp, 2022-05-21:
Hello Lee Yang,
Upon review, I see the problem.
In the config/GALILRELEASE file there is this entry for the GALIL macro:
GALIL=/opt/epics/modules/Galil-3-0/3-6
but according to the build log, the Galil EPICS driver is located in the folder below.
          /opt/epics/modules/motorGalil/Galil-3-0/3-6
Please alter the GALILRELEASE file so that the GALIL macro points to a folder that does exist, and contains the Galil EPICS driver
Recommend the GALIL macro entry in the configure/RELEASE file be removed as it's redundant.
Best of luck.
Kind regards,
Mark.

LeeYangLBLBCS, 2022-05-21:
I'm very sorry for the error, and thank you very much for your help.

closed 2022-05-21 by LeeYangLBLBCS