motorapp/Galil #16: Home failed with auto-generated code

Huijuan-Xu, 2022-06-27:
The auto-generated home code doesn't work well when there's home switch. The motor stops at forward limit switch/reverse limit switch when running HOMF/HOMR and give time-out error. Current code has fixed value for hswact and hswiact while the state of the home (_HM) switch depends on the value set for _CN1. When _CN1 is set to -1, _HM will show 0 when inactive and 1 when active. When _CN1 is set to 1, _HM will show 1 when inactive and 0 when active.

motorapp, 2022-06-28:
Hi Huijuan-Xu,
From your message, I see you know Galil low level commands.

The _CN1 entangles 2 concepts together, they are:
1.  Home switch polarity
2.  Motor direction used in HM, FE command based on home switch input state
With the Galil EPICS driver these concepts are no longer bound together, by avoiding use of HM, FE commands for the auto-generated home routines.  However, the HM, FE commands can be used in custom home routines when appropriate.
With the Galil EPICS driver _CN1=-1=Normally open home switch type, whilst _CN1=1=Normally closed home switch type.  Factoring in the switch state when inactive/active for these switch types then hswact is always 0 and hswiact is always 1.  Expressing the same idea another way, it is assumed when _CN1 is changed, so is the home switch type such that hswact is always 0 and hswiact is always 1. 

The PV $(P)HOMETYPE_CMD, where P=DMC01: by default, can be used to change _CN1.

The iocShell command GalilReplaceHomeCode can be used to specify a custom home routine.  The file iocBoot/iocGalilTest/galil.cmd describes the GalilReplaceHomeCode command.

Could you please describe your motor, encoder, limits and home switch setup?
Could you please describe how you would like the motor to home?

Kind regards,
Mark

Huijuan-Xu, 2022-06-28:
Hi Mark,
Thanks for your response.
If I understand correctly, the _CN1 should be set to 1 regardless how it is
configured in order to make the auto-generated home code work. That means
the PV $(P)HOMETYPE_CMD should be set to "NC". I tried that but still got
the same result. Every time when a 'HOMF'/'HOMR' is issued, the motor
passes through HOME, stops at FLS/RLS and gives a timed-out error. By the
way, I assume the home position is found when _HM?=hswact?. In this case,
hjog should be set to 3. Is that correct?

The test stand includes a Galil DMC-4020, encoder(Renishaw RGH24H50D30A),
motor (Airex LMC-P12-41S024N SP55),  amplifier (Trust Automations
TA310-H01). There are 3 non-contact switches (OHS3140U) wired in the test
stand. They are wired to FLSA, HOMA and RLSA respectively. Please take a
look at the attached images.

There's no special requirement for my homing program. If 'HOMF' is issued,
the motor should move and stop at HOMA, or stop at FLS and return to HOMA
position depending on the starting position. Similar to the 'HOMR'. This
should be a common requirement of the home operation and should work with
the auto-generated code. Please let me know if there's anything wrong.

Thanks,
Huijuan

motorapp, 2022-06-29:
Hi Huijuan, 
Thanks for describing the hardware in use.  I can't see the attachments.  It might be easiest to email me at padmoz@tpg.com.au.

To answer the questions briefly.
I assume the home position is found when _HM?=hswact?.
**Yes.**
In this case, hjog should be set to 3. Is that correct?
**Not necessarily, it depends on software configuration of Galil EPICS driver.**

I have further questions:
Did you configure the IOC using the Galil EPICS driver?, OR
Is the example IOC that comes with the driver being used instead?
Has the GalilCreateAxis command for axis A been altered to use home switch, instead of limit switches, for home? (parameter 3)
What GUI are you using to interact with Galil EPICS driver?
Given hardware description, homing to home switch is not the most accurate option available.  Why is it desired to home to the home switch? (eg. near beam center location, etc)

It would be great to see the wiring images for the FLS, RLS (reverse limit switch) and home switch.

I look forward to your email.
Kind regards,
Mark

closed 2022-08-17 by motorapp