motorapp/Galil #17: Loss on information about homedX variables upon IOC restart

guijmontagner, 2022-10-28:
This issue is aimed to report a detected misbehaviour of the Galil-3-0 IOC which consists on the loss of information about homedA,homedB,homedC... variables (which are stored on a Galil controller) values after IOC initialization.

These variables contain the status of the sucessfull completion of the last commanded homing routine, and are useful to determine positioning safety of a system on a synchroton facility.

The detected  behaviour is: 
1) The system controlled by the Galil controller is already homed, and the status (as obtained via use of Galil IDE "GDK", using the terminal tool) of "homedA" and "homedB" variables - which contain information about homing completion on axes A and B of the Galil - is currently equal to "1", which means both axes have already been homed with sucess. 
2) The IOC is not running. 
3) Then, the IOC is started via its ".cmd" file, as usual.
4) Using GDK terminal again, the value of those "homedA" and "homedB" variables is obtained, and is found to be now equal to "0"

This behaviour is unwanted, since restarting an IOC shouldn't affect the status of a homing command for any axes, as this is an information that regards only what happens with encoder readings/stepper motor pulses countings, which will only be affected by shutting off the Galil controller or artificially changing these encoder/motor associated position values.

Given this, it is provided (at the end of this post) an image of the messages output from EPICS Galil IOC at the time, and it is shown errors have happened, particularly the one that reads "Error setting variables".

The questions are: 
- Can this error message imply the Galil-3-0 IOC is resetting the "homedA" and "homedB" variables at startup, causing the problem?
- If so, is this related to EEPROM burn failure?
- If so, is there a way to obtain a status bit from _GalilStartController_ function (which is issued by the IOC associated with the Galil controller in question) so to modify the IOC initialization parameters and code specific to the system in question, and customize the IOC so it can issue additional attempts on burning code and variables whenever the first attempt fails, or even issue warnings to end users informing them the controller has not been setup successfully by the IOC for its intended operation?

See below the mentioned image with the evidences for the detected behaviour

[image]

motorapp, 2022-10-29:
Hello,
The reason for the error is related to bad communication happening with the controller.

Inspecting the provided logs, the firmware on the controller is too old.
Please update the firmware to version 1.2F (preferred) or above, then try again.

motorapp, 2022-11-03:
Hello,
The cause of the reported issue is communication problems with the controller that is caused by firmware that is too old.  A minimum of firmware 1.2F is required for that controller model.

Providing further information below.
1.  Yes, the Galil driver can set homedA-H to 0 during the setup of the required variables on the controller, but the process is not completed successfully in the reported case due to communication issues, due to firmware issues.  The driver will decide to set up the variables on the controller if any of the following is true: 
      A).  The user has specified burn program to EEPROM in GalilStartController call AND the program specified was downloaded to the controller successfully because it was different from what was on the controller.
     B)  The controller doesn't know the homedA-H variable
     C)  The controller doesn't know the commutation initialized variable cinitA-H
To set up the variables on the controller the driver does the following.
     A)  Backup value of cinitA-H, backup value of homedA-H
     B)  Set variables to desired power on value (homedA-H=0, CINITA-H=0)
     C)  Burn variables to EEPROM
     D)  Restore cinitA-H, homedA-H value that was captured in step A) above
2.  The firmware below 1.2F on this controller has many known bugs.  The minimum firmware is 1.2F.  Firmware 1.2F assists greatly with an issue where ethernet traffic locks up the controller causing the red LED to come on.  However, specific network configuration is most often required too.  Please refer to the networking guide in the git repository documentation folder.
3.  No.  With given firmware 1.1d additional attempts will also fail.
     The PV $(P)START_STATUS can be used to see if the code specified started ok.  It is checked only at IOC start.
     The driver only sets the controller program and variables if necessary.  
     Once the specified program and variables are burnt to EEPROM, the driver won't do the program and variable set 
     up again, until it's altered by the operator.

If you'd like to discuss your project further, please email me at padmoz@tpg.com.au

closed 2022-11-03 by motorapp