motorapp/Galil #18: help with the default st.cmd script

LeeYangLBLBCS, 2022-11-09:
the default start up script contains references to a second controller "rio", in additional to the first controller "Galil".
https://github.com/motorapp/Galil-3-0/blob/master/3-6/iocBoot/iocGalilTest/galil.cmd
I wonder what the "rio" is meant to be.
Is it just another Galil controller at a different IP address?
I happen to have two Galil controllers. Should I modified "galil.com" so make "rio" the 2nd controller?

motorapp, 2022-11-10:
Galil products include motor controllers and PLC like devices for remote I/O (RIO).  The RIO devices provide analog and digital IO across an ethernet or serial connection, they don't provide motor control.
The Galil EPICS driver supports both Galil motor controllers and RIO controllers.  A single IOC can have both motor controllers and RIO devices.  The number of Galil controllers (motor controller or RIO) an IOC can support depends on the IOC hardware and ethernet connection.
The "Galil" and "RIO" strings appear in galil.com as the GalilCreateController 1st parameter.  GalilCreateController first parameter is an arbitrarily assigned asynPort name (ie. "Galil", "RIO").  The port name used in the database substitution files located in the GalilTestApp/Db folder (ie. macro PORT), must match the asynPort name used in galil.com/GalilCreateController parameter 1.
Adding a second controller takes more steps but could include replacing "RIO" with something like "Galil2" or "GalilB".
Check out the configuration guide at the link below:
https://github.com/motorapp/Galil-3-0/blob/master/3-6/documentation/Galil%20EPICS%20configuration%20guide.pdf

In general, to add a controller the following things must be altered.
1.  IOC start scripts (st.cmd, galil.cmd)
2.  Database substitution files (GalilTestApp/Db *.substitutions files)
3.  Autosave request files (iocBoot/iocGalilTest *.req files)
4.  Duplicate and alter duplicate display start scripts (MEDM or Qt) (/3-6 start_dmc_screen.sh and 
     start_qedmc_screen.sh)

If you have questions afterward, let me know.

LeeYangLBLBCS, 2022-11-10:
thanks for your help.
I opted to make a copy of the original st.cmd file, call it e.g. st2.cmd
then replaced all references to "DMC01:" by "DMC02:", and "Galil" by "Galil2" in all related files, (including all the substitution files in Db).
That almost worked, except one of the axes (Axis E).
The IOC startup script keeps changing the motor type value from 2 to -1 immediately after the IOC starts.
All motors are steppers, without encoders. So they should all be type 2.
I'm attaching the st2.cmd for reference below.

=========================file st2.cmd=======================
#!/opt/epics/modules/motorGalil/Galil-3-0/3-6/bin/linux-x86_64/GalilTest

< envPaths

#cd ${TOP}

## Register all support components
dbLoadDatabase("${TOP}/dbd/GalilTest.dbd",0,0)
GalilTest_registerRecordDeviceDriver(pdbbase)

#cd ${TOP}/iocBoot/${IOC}

### Scan-support software
# crate-resident scan.  This executes 1D, 2D, 3D, and 4D scans, and caches
# 1D data, but it doesn't store anything to disk.  (See 'saveData' below for that.)
#dbLoadRecords("$(SSCAN)/sscanApp/Db/standardScans.db","P=IOC02:,MAXPTS1=8000,MAXPTS2=1000,MAXPTS3=10,MAXPTS4=10,MAXPTSH=8000")
#dbLoadRecords("$(SSCAN)/sscanApp/Db/saveData.db","P=IOC02:")

# Configure an example controller
< galil2.cmd

#< autosave.cmd

# Start the IOC
iocInit()

# Initialize saveData for step scans
#saveData_Init("saveData.req", "P=IOC02:")

# Save motor positions every 5 seconds
#create_monitor_set("GalilTest_positions.req", 5,"P=DMC02:")
# Save motor settings every 30 seconds
#create_monitor_set("GalilTest_settings.req", 30,"IOC=IOC02:,P1=DMC02:, P2=RIO02:")
# Save kinematics every 30 seconds
#create_monitor_set("GalilTest_kinematics.req", 30,"P1=DMC02:")

# end

motorapp, 2022-11-11:
From the list below, which is closest to the desired objective?
A)  Two Galil motor controllers on a single IOC, or
B)  Two IOC's with 1 motor controller on each.  PV names can be the same, or
C)  Two IOC's with 1 motor controller on each.  PV names must be different

The steps taken so far align best with options B and C above.
The provided st2.cmd is only part of the information that would be required to assist.

The reported issue is not a problem with the Galil EPICS driver.  I will close this issue in the next day or so.
Please email me at padmoz@tpg.com.au for further assistance.  I am happy to assist further.
Kind regards,
Mark

Section 2.4 onward has the information you are seeking.
https://github.com/motorapp/Galil-3-0/blob/master/3-6/documentation/Galil%20EPICS%20configuration%20guide.pdf

LeeYangLBLBCS, 2022-11-11:
I just noticed the default values in this file for "MTRTYPE" look wrong galil_motor_extras.substitutions:
file "$(GALIL)/GalilSup/Db/galil_motor_extras.template"
{
pattern
{   P,        M,    PORT,    ADDR, PREC, SCAN,       MTRTYPE, MTRON, EGU  }
 {  "DMC02:", "A",  "Galil2", 0,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "B",  "Galil2", 1,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "C",  "Galil2", 2,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "D",  "Galil2", 3,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "E",  "Galil2", 4,    3,    "Passive",  "1",     "0",   "mm" }
 {  "DMC02:", "F",  "Galil2", 5,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "G",  "Galil2", 6,    3,    "Passive",  "3",     "0",   "mm" }
 {  "DMC02:", "H",  "Galil2", 7,    3,    "Passive",  "3",     "0",   "mm" }

I changed all rows for "MTRYPE" to "2", assuming it is for "MT" in Galil controller motor settings.

This appeared to have fixed the problem with Axis E, which had value of 1.
Though the MT values in the controller become "-2", not "2".
=========Galil tool output========
:MT?,?,?,?,?,?,?,?
 -2.0, -2.0, -2.0, -2.0, -2.0, -2.0, -2.0, -2.0
Since either +2 or -2 are steppers, so the motors seem to move.
But I'd like to understand where the extra - sign come from.
I also tried to enter "-2" in this file for MTRYPE - but that didn't result in +2, rather a -1 in MT value.

LeeYangLBLBCS, 2022-11-11:
Sorry for not reading the comments in "galil_motor_extras.substitutions":
# MTRTYPE - motor type =
#               0 - Servo
#               1 - Reverse servo
#               2 - High active stepper
#               3 - Low active stepper
#               4 - Reverse high active stepper
#               5 - Reverse low active stepper
# MTRON - motor off 0, motor on 1
that explained all the problems.
thanks very much.

closed 2022-11-11 by LeeYangLBLBCS