motorapp/Galil #21: Galil controller stuck suddenly

PeiyuQuan, 2024-06-08:
I found a new issue; I am not sure what cause it.
I found galil_dmc_4183 be stuck several times during last beamtime, it shows "GalilAxis::move begin failure axis B", In fact, all channels couldn't work, I must restart controller to fix it or both restart controller and restart computer to fix it.
Because restart controller wasn't enough sometimes.
If I only restart controller and run IOC again. The error report shows: "Asynchronous UDP failed, switching to TCP synchronous" and Communication status show Error and Controller start statue also shows error.
I must restart PC sometimes. 
Does any people know what happened it is? Thank you
[image]
[image]

motorapp, 2024-06-11:
Hi,
Reading the description of the problem, it sounds like a networking issue.  Please check the firmware on the controller.  To obtain the firmware version, check the DMC-Controller screen model at the top of the DMC controller screen, or connect with GalilTools or GDK and press Control R, then hold control and press V, then press enter.  The minimum required firmware for the DMC-4103 Series controller is 1.2F.  New firmware for the DMC-4103 Series controller can be obtained from Galil at this weblink https://www.galil.com/downloads/firmware.  If you find the controller firmware is at or newer than 1.2F, please let me know.
I also recommend reading the networking Guide located here: https://github.com/motorapp/Galil/tree/master/documentation
Please let me know how it progresses.
Kind regards,
Mark

PeiyuQuan, 2024-06-17:
Hello Mark

Thank you for your replying, I am sorry to answer you so late, I am out of town last week. I just checked my controller, its firmware is DMC4183s01a. I don't know how many years it wasn't updated! need to download the latest firmware for it. If it has other result, I will let you know. Thank you.

closed 2024-06-18 by motorapp