motorapp/Galil #22: Latest version does not work with DMC2182.

AbdallaDalleh, 2024-08-13:
Hi

We tried installing the latest version of the Galil driver to work with a DMC2182, the IOC shell showed this error:

`Transferring code to model DMC2182 Rev 1.0v, address x.x.x.x`
`Error downloading code model DMC2182 Rev 1.0v, address x.x.x.x`

This will result in a control start status of "Error". This is a serious issue because soon we will upgrade our motion IOC servers, there is no indication to which commit of the v3.6 that works with the DMC2182 or anything prior to the DMC-4000. Any insight is appreciated.

motorapp, 2024-08-13:
Hi,
The Galil driver is tested regularly on a 2183 amongst other models.  I am unable to reproduce the error on the test 2183 1.0V hardware.  Currently, I don't have other reports of network issues on the 2183 model.  The 1.0V firmware is fine.
To isolate the issue, I recommend the following:
1.  Start with easiest.  Reboot the controller and try again.  Determine how reproducible the error is on that controller.
2. Ensure the operating system/network MTU is 1500 bytes or less
3. Connect with GalilTools or GDK to test network connectivity
4. Resort to direct network crossover cable for testing
5. Try another 21x3 controller
6. Read the networking guide
7. Ensure only the driver is attempting to communicate with the controller

The 2183 controller when heated over 70 degrees will also have network issues that may be permanent.

In case the issue becomes complex, feel free to email me at padmoz@tpg.com.au
Please let me know what you find.
Kind regards,
Mark

closed 2024-08-24 by motorapp

AbdallaDalleh, 2024-08-28:
Hi Mark

It appears the root cause was improper format for the cmd function `GalilCreateAxis`, in the new version it has 4 parameters:

1. Port name
2. axis
3. Motor interlock digital port number
4. Interlock switch type.

While the old version had an additional 3rd parameter `limits_as_home`. In the current `cmd` file we have

`GalilCreateAxis("Galil", "A", 1, "", 1)`

Which on the new version it passes 1 as an interlock port number, thus generating extra code in the DMC in the function `GalilController::gen_motor_enables_code`. This is not itself the issue, the issue is in the aforementioned function which enables interlock on all axes on a single program line, the same DMC was downloaded to the controller using Galil tools and showed that there is a line longer than 80 characters which is the line written by this function. A fix was tested and the generated DMC works fine. I think a more general fix would in the `programDownload` function which fixes the entire program by shortening any line longer than 80 characters before downloading, what do you think?

I would like to raise an important issue here, the versioning structure of the Galil repo is becoming really confusing and creating more issues in IOCs development and deployment. Therefore, I propose the following:

1. Work more on following the semantic versioning rules. (https://semver.org/) 
2. Galil 3.6 is not 3.6 any more, looking at the change log in the newest tag, we are not talking about 85 patches, these are minor features and I am sure some are major features that might break compatibility. I would assume this change log would raise the version number at least to v3.10
3. Again in the change log we are not talking about 85 patches and thus setting the patch version number to 85 does not make any sense.
4. v3.4 or v3.5 to v3.6 was not a minor, backward compatible feature, it was a major backward-incompatible feature. For example complete PVs renaming which broke existing GUIs, tools and scripts when we upgraded. In this case, v3.6 should be at least v4.0

Thank you for your understanding!
Abdalla.