motorapp/Galil #26: autosave directory should be created automatically on cloning

MarkRivers, 2024-12-28:
It is convenient to have the iocBoot/iocGalilTest/autosave directory created automatically when cloning.  The changes to .gitignore and the addition of README.txt to the autosave directory do this. (#25)

.gitignore also added RELEASE*.local* for local release files.

.gitignore also added the autogenerated .trj files.

motorapp, 2025-01-01:
Is this a site specific requirement?  I'm happy to merge it either way.  Could you briefly describe the requirement that drives this addition?  If it's just a convenience item, I'll still merge it.  I don't see this folder in other example IOC's, just curious.  Thanks.

MarkRivers, 2025-01-01:
Without this change then if one clones Galil, builds, and runs the example IOC there will be errors because the autosave.cmd tells it to use the /autosave subdirectory:
https://github.com/motorapp/Galil/blob/97872c534d3e7016cec1278f7791c6e2426c80df/iocBoot/iocGalilTest/autosave.cmd#L20

but that directory won't exist.

These are all the autosave directories in my synApps tree that have autosave/README* files in them.  So it is a common practice.  It is not done in the motor/modules example IOCs, because those are not built with autosave at all.
```
[epics@corvette support]$ find . -name 'README*' | grep autosave
./quadEM-9-4/iocBoot/iocAH401B/autosave/README
./quadEM-9-4/iocBoot/iocAH501/autosave/README
./quadEM-9-4/iocBoot/iocAPS_EM/autosave/README
./quadEM-9-4/iocBoot/iocTetrAMM/autosave/README
./quadEM-9-4/iocBoot/iocNSLS2_EM/autosave/README
./quadEM-9-4/iocBoot/iocNSLS_EM/autosave/README
./quadEM-9-4/iocBoot/iocNSLS2_IC/autosave/README
./dante-1-1/iocBoot/iocDante1/autosave/README
./dante-1-1/iocBoot/iocDante8/autosave/README
./measComp-4-2/iocBoot/iocTC32/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB1208/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB1608G/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB2408/autosave/README.txt
./measComp-4-2/iocBoot/iocUSBCTR/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB231/autosave/README.txt
./measComp-4-2/iocBoot/iocE1608/autosave/README.txt
./measComp-4-2/iocBoot/iocETC/autosave/README.txt
./measComp-4-2/iocBoot/iocEDIO24/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB1608G-2AO/autosave/README.txt
./measComp-4-2/iocBoot/iocUSBTEMP-AI/autosave/README.txt
./measComp-4-2/iocBoot/iocUSBTEMP/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB3105/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB1608HS-2AO/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB1808/autosave/README.txt
./measComp-4-2/iocBoot/iocUSB3104/autosave/README.txt
./measComp-4-2/iocBoot/iocUSBSSR/autosave/README.txt
./measComp-4-2/iocBoot/iocUSBERB24/autosave/README.txt
./tomoscan/iocBoot/iocTomoScan_13BM_MCS/autosave/README
./tomoscan/iocBoot/iocTomoScan_13BM_PSO/autosave/README
./tomoscan/iocBoot/iocTomoScan_7BM/autosave/README
./tomoscan/iocBoot/iocTomoScanStream_2BMA/autosave/README
./tomoscan/iocBoot/iocTomoScanStream_2BMB/autosave/README
./tomoscan/iocBoot/iocTomoScan_2BMA/autosave/README
./tomoscan/iocBoot/iocTomoScan_2BMB/autosave/README
./tomoscan/iocBoot/iocTomoScan_2BMB_STEP/autosave/README
./tomoscan/iocBoot/iocTomoScan_32ID/autosave/README
./tomoscan/iocBoot/iocTomoScan_32ID_STEP/autosave/README
./tomoscan/iocBoot/iocTomoScan_6BM_STEP/autosave/README
./tomoscan/iocBoot/iocTomoScanStream_32ID/autosave/README
./tomoscan/iocBoot/iocTomoScanStream_7BM/autosave/README
./labjack-3-0/iocBoot/iocLabJack_T4/autosave/README
./labjack-3-0/iocBoot/iocLabJack_T7/autosave/README
./labjack-3-0/iocBoot/iocLabJack_T7Pro/autosave/README
./labjack-3-0/iocBoot/iocLabJack_T8/autosave/README
./mca-7-9/iocBoot/iocDarwin/autosave/README
./autosave-5-10-2/README.md
./dxp-6-0/iocBoot/iocDXP2X/autosave/README
./dxp-6-0/iocBoot/iocMercury/autosave/README
./dxp-6-0/iocBoot/iocSaturn/autosave/README
./dxp-6-0/iocBoot/iocXMAP/autosave/README
./dxpSITORO-1-3/iocBoot/iocFalconX4/autosave/README
./dxpSITORO-1-3/iocBoot/iocFalconX1/autosave/README
./dxpSITORO-1-3/iocBoot/iocFalconX8/autosave/README
./areaDetector-3-13/aravisGigE/iocs/aravisGigEIOC/iocBoot/iocAravisGigE/autosave/README
./areaDetector-3-13/ADmarCCD/iocs/marCCDIOC/iocBoot/iocMARCCD/autosave/README
./areaDetector-3-13/ADmar345/iocs/mar345IOC/iocBoot/iocMAR345/autosave/README
./areaDetector-3-13/ADURL/iocs/urlIOC/iocBoot/iocURLDriver/autosave/README
./areaDetector-3-13/ADRoper/iocs/roperIOC/iocBoot/iocRoper/autosave/README
./areaDetector-3-13/ADProsilica/iocs/prosilicaIOC/iocBoot/iocProsilica/autosave/README
./areaDetector-3-13/ADPointGrey/iocs/pointGreyIOC/iocBoot/iocPointGrey/autosave/README
./areaDetector-3-13/ADPointGrey/iocs/pointGreyIOC/iocBoot/iocPointGreyLVP/autosave/README
./areaDetector-3-13/ADPixirad/iocs/pixiradIOC/iocBoot/iocPixirad/autosave/README
[... 5 lines cut ...]
./areaDetector-3-13/ADFastCCD/iocs/FastCCDIOC/iocBoot/iocFastCCD/autosave/README
./areaDetector-3-13/ADDexela/iocs/dexelaIOC/iocBoot/iocDexela/autosave/README
./areaDetector-3-13/ADBruker/iocs/brukerIOC/iocBoot/iocBruker/autosave/README
./areaDetector-3-13/ADAndor3/iocs/andor3IOC/iocBoot/iocAndor3/autosave/README
./areaDetector-3-13/ADAndor/iocs/andorIOC/iocBoot/iocAndor/autosave/README
./areaDetector-3-13/ADADSC/iocs/adscIOC/iocBoot/iocAdsc/autosave/README
./areaDetector-3-13/ADMerlin_old/iocs/merlinIOC/iocBoot/iocMerlin/autosave/README
./areaDetector-3-13/ADMerlin/iocs/merlinIOC/iocBoot/iocMerlin/autosave/README
./areaDetector-3-13/ADCSimDetector/iocs/ADCSimDetectorIOC/iocBoot/iocADCSimDetector/autosave/README
./areaDetector-3-13/ADLambda/iocs/LambdaIOC/iocBoot/iocLambda/autosave/README
./areaDetector-3-13/ADPCO/iocs/pcoIOC/iocBoot/iocPCO/autosave/README
./areaDetector-3-13/ADSimDetector/iocs/simDetectorIOC/iocBoot/iocHDF5Test/autosave/README
./areaDetector-3-13/ADSimDetector/iocs/simDetectorIOC/iocBoot/iocSimDetector/autosave/README
./areaDetector-3-13/NDDriverStdArrays/iocs/NDDriverStdArraysIOC/iocBoot/iocNDDriverStdArrays/autosave/README
./areaDetector-3-13/pvaDriver/iocs/pvaDriverIOC/iocBoot/iocPvaDriver/autosave/README
./areaDetector-3-13/ADEiger/iocs/eigerIOC/iocBoot/iocEiger1/autosave/README
[... cut ...]
./areaDetector-3-13/ADPylon/iocs/pylonIOC/iocBoot/iocPylon/autosave/README
./areaDetector-3-13/ADVimba/iocs/vimbaIOC/iocBoot/iocVimba/autosave/README
./areaDetector-3-13/ADSpinnaker/iocs/spinnakerIOC/iocBoot/iocSpinnaker/autosave/README
./areaDetector-3-13/ADEuresys/iocs/EuresysIOC/iocBoot/iocAdimec/autosave/README
./areaDetector-3-13/ADEuresys/iocs/EuresysIOC/iocBoot/iocMikrotron/autosave/README
./areaDetector-3-13/ADEuresys/iocs/EuresysIOC/iocBoot/iocViewWorks/autosave/README
./areaDetector-3-13/ADEuresys/iocs/EuresysIOC/iocBoot/iocOrca/autosave/README
```

motorapp, 2025-01-02:
Thanks for taking the time to provide further information, I am grateful.  Will merge as requested.

motorapp, 2025-01-13:
Merged manually

closed 2025-01-13 by motorapp