motorapp/Galil #27: .dmc files should be installed in the /db directory

MarkRivers, 2024-12-28:
.dmc files should be installed in the /db directory so other applications can find them without needed to know the Galil tree.

GalilSup/Db/Makefile is changed to do this in #25.

motorapp, 2025-01-01:
Does the site (GSECARS) use the dmc files provided in GalilSup/Db to construct Galil microcode?

The template mechanism of defining the Galil microcode is supported, but I encourage folks to try the inbuilt home code generator as it does receive a greater level of support/testing.  Also, some of the features in the motor_extras database may not function when using the dmc templates provided in the repository.  Could you let me know in the case that the site has tried the home code generator and found it not meeting requirements?  In this case, the information feedback could help me.

Using the code generator, folks can use the GalilReplaceHomeCode, and GalilAddCode iocShell functions to put their own code in place.  When using custom code, the galil_userdef_records.* allows connection between the micro code and records.  To establish custom micro-code/record interface here are the steps:

1.  In galil_userdef_records.substitutions define a record that connects to a galil microcode variable. (For example variable = begin).  Set the SCAN field to "I/O Intr"
2. Here is sample Galil micro-code that will interface with the variable created in step 1. above

IF(begin=1)
// REM enter this block only if begin=1 (can be set by record created in  galil_userdef_records.substitutions)
// REM custom code begin
// REM custom code body
// REM custom code end
// REM custom code completed so set begin to 0 accordingly
begin=0
// REM Tell EPICS the variable begin new value
MG "begin", begin
ENDIF

3.  The driver knows a record has been defined to monitor the variable "begin", and the message will be routed to the correct record.
4. If the driver can't find the record, the message is sent to $(P)ERROR_MON PV that is normally displayed on the bottom of the DMC screen.

motorapp, 2025-01-01:
Regarding Galil micro-code.  In my experience the 4080 is between 5x and 10x faster than the 4183.  Aside from the micro-code performance, the motion performance is similar, mostly.

MarkRivers, 2025-01-01:
> Does the site (GSECARS) use the dmc files provided in GalilSup/Db to construct Galil microcode?

No, we have not done that yet.

>  Could you let me know in the case that the site has tried the home code generator and found it not meeting requirements?

I plan to experiment with microcode soon.  I'll let you know if I have any suggestions.

motorapp, 2025-01-02:
Thanks for the update.  Always grateful for your help/suggestions Mark.)

motorapp, 2025-01-13:
Merged manually

closed 2025-01-13 by motorapp