motorapp/Galil #29: Provide default values for many macros in galil_motor-v6-10up.template

MarkRivers, 2024-12-28:
galil_motor-v6-10up.template is modified to provide default zero values for many macro arguments.  This allows this database to be loaded with fewer arguments when desired.

MarkRivers, 2024-12-28:
Fixed in #25

motorapp, 2025-01-01:
Thanks for this update, addition.  I've merged most of the changes in this case.  Excepting the default value of MRES=0, and ERES=0.  Please reach out if any concerns arise.

MarkRivers, 2025-01-01:
> I've merged most of the changes in this case. Excepting the default value of MRES=0, and ERES=0.

I want to keep MRES=0 and ERES=0 defaults.  
- Many sites specify SREV and UREV rather than MRES in the .substitutions files.   The motor record then computes MRES=UREV/SREV.  We don't want to have to specify MRES=0 for all records in the substitutions file.
- Nearly all of our Galil motors are open-loop steppers, so we don't specify ERES in the substitutions files.  We don't want to have to add ERES=0 to the substitutions file.

I don't see any harm in have ERES=0 and MRES=0 defaults.  It won't break anything for sites that were passing those anyway.

motorapp, 2025-01-02:
Thanks for taking the time to provide further information, I'm grateful.  Will merge as requested.

motorapp, 2025-01-13:
Merged manually.

closed 2025-01-13 by motorapp