motorapp/Galil #33: Bug fixes and enhancements in GalilController.cpp

MarkRivers, 2024-12-28:
There are bugs in GalilController.cpp around line 4300.   The existing code was this:
```
           mesg = pAxis->axisName_ + " wrong limit protect disabled at end(s) with disabled limits";
```
This is incorrect because it is using the C++ string concatenation but the first argument is a char, not a string.  The result was corrupted message strings, even printing parts of other strings.

This could be fixed by changing it to string(1, pAxis->axisName_).  However, I changed it to provide the address_ and Axis string, which was already done for some messages in that area but not all.  In the new code the string() constructor is actually not needed because the  + operator will now do it, since there is a string to the left.  But it is clearer to include the string() constructor.  The new code is
```
           mesg = address_ + " Axis " + string(1, pAxis->axisName_) + " wrong limit protect disabled at end(s) with disabled limits";
```
Other changes in this file are:
- Include <ciso646>on Windows to allow use of "and" rather than &&.
- Provide more information in the report() function.  This was previously commented out, and I fixed it up.
- Print error message for unsupported amplifier models.
-

MarkRivers, 2024-12-28:
Fixed in #25

MarkRivers, 2024-12-29:
Added new commit that improves amplifier handling.
- Handle servo amps when callbacks needed.
- Handle case of no amplifier.
- Change from long if/else block to switch statements which are more readable.

motorapp, 2025-01-02:
Thanks for the bug fixes and contributions, I am very grateful.
All will be merged excepting the include file on windows to allow use of the "and" keyword.
Please reach out if any concerns arise.

MarkRivers, 2025-01-02:
> All will be merged excepting the include file on windows to allow use of the "and" keyword.

I just tested on Windows with that line commented out, and it works fine.  I believe I added that because an earlier version of the driver used "and" or "or" words, rather than && or ||.  Is that correct, were those removed?

motorapp, 2025-01-03:
Yes, that is correct.  It seems the keyword "and" was accepted in C++20 and will appear in the header "version".

closed 2025-01-13 by motorapp