motorapp/Galil #34: Allow Galil module to be build without error on systems that don't have C++11

MarkRivers, 2024-12-28:
We want to build IOCs for many OS using the same set of modules.  Some of the OS are old and don't support C++11, e.g. CentOS 6 and RHEL 6.  

This cannot be done by just skipping the build in the Makefile, because the dbd that references the driver is in O.Common, so the code must exist.  I did the following:
- Added GalilDummy.cpp to create the registrar entry.
- Changed src/Makefile to build the real driver if HAVE_C++11 is true (the default) and the dummy driver if it is false.
- Added CONFIG_SITE.Common.$(EPICS_HOST_ARCH) for a few systems that don't have C++11.

MarkRivers, 2024-12-28:
Fixed in #25.

motorapp, 2025-01-01:
I'll merge this change.

In case running C++11 compiler on older CentOS 6 is of interest, devtoolset-7 is recommended.
Attached is 1 script file to obtain devtoolset-7 for CentOS 6.10 from cloudlinux website, and 1 file listing tips and the steps I used.
[devtoolset7_install.sh.txt](https://github.com/user-attachments/files/18284499/devtoolset7_install.sh.txt)
[DevToolSet7Help.txt](https://github.com/user-attachments/files/18284500/DevToolSet7Help.txt)

closed 2025-01-13 by motorapp