motorapp/Galil #35: Strange performance problem when single-stepping an open-loop stepper motor

MarkRivers, 2024-12-30:
I am seeing a very strange behavior when execute a set of single-step motor record moves from a script.

Configuration:
-	EPICS base 7.0.7
-	EPICS motor record, R7-3-1
-	Galil DMC-4183 motor controller
-	Galil driver V4-0-02 with minor changes
-	Open-loop stepper motor with MRES=.00125
-	Backlash distance BDST = 0.

I have a script that repeatedly moves the motor in single steps by incrementing or decrementing the motor record RVAL field by 1.
The script writes to the RVAL field with ca_put_callback, so it returns as soon as the move is complete, and then starts the next move.

In principle these moves should be independent of the motor record VELO and ACCL fields, since it is only moving a single step.

I observe the behavior by monitoring the done moving (DMOV) field, which is 0 when the motor is moving  and 1 when a move is complete.

If I set the VELO field to 1.0 (800 steps/s) this is the camonitor result.

```
(base) [epics@corvette motor]$ camonitor -tc 13BMD:m29.DMOV
13BMD:m29.DMOV                 (2024-12-30 13:17:55.364647) 1
13BMD:m29.DMOV                 (2024-12-30 13:17:55.364896) 0
13BMD:m29.DMOV                 (2024-12-30 13:17:55.403801) 1
13BMD:m29.DMOV                 (2024-12-30 13:17:55.404055) 0
13BMD:m29.DMOV                 (2024-12-30 13:17:55.442796) 1
13BMD:m29.DMOV                 (2024-12-30 13:17:55.443167) 0
13BMD:m29.DMOV                 (2024-12-30 13:17:55.481823) 1
13BMD:m29.DMOV                 (2024-12-30 13:17:55.482166) 0
13BMD:m29.DMOV                 (2024-12-30 13:17:55.521032) 1
```
The time to go from 1 (done) to 0 (moving) is less than 1 ms.  This is the time after motion done callback for the script to begin the next move.

The time to go from 0 (moving) to 1 (done) is about 40 ms.  This is the time for the driver to send Ethernet command to the controller to begin the motion, for the move to complete, and for the driver to get the completion status from the controller, and to issue the CA monitors.

These timings make sense to me.

However, if I increase the VELO field from 1.0 to 1.5 (1200 steps/s) then this is the camonitor result:

```
(base) [epics@corvette motor]$ camonitor -tc 13BMD:m29.DMOV
13BMD:m29.DMOV                 (2024-12-30 13:26:33.994636) 1
13BMD:m29.DMOV                 (2024-12-30 13:26:33.995166) 0
13BMD:m29.DMOV                 (2024-12-30 13:26:34.502491) 1
13BMD:m29.DMOV                 (2024-12-30 13:26:34.502939) 0
13BMD:m29.DMOV                 (2024-12-30 13:26:35.010406) 1
13BMD:m29.DMOV                 (2024-12-30 13:26:35.010706) 0
13BMD:m29.DMOV                 (2024-12-30 13:26:35.518281) 1
13BMD:m29.DMOV                 (2024-12-30 13:26:35.518503) 0
13BMD:m29.DMOV                 (2024-12-30 13:26:36.026216) 1
13BMD:m29.DMOV                 (2024-12-30 13:26:36.026736) 0
```
The 1 to 0 time is still less than 1 ms.

But the 0 to 1 time is now about 508 ms.  It has increased by more than a factor of 12 compared to the 40 ms when VELO=1.0.

This makes no sense to me.  Why does changing the VELO field affect the time for a single-step move?

This is a screenshot of the motorx_all screen when the problem occurs:
[image]

This is the galil_motor_extras screen:
[image]

Is this a problem with the motor record, or with the Galil driver?

MarkRivers, 2024-12-31:
I can reproduce this behavior using the SSCAN record to increment the RVAL field, rather than a script.  

This is the configuration of the SSCAN record.  It is incrementing RVAL by 1, with 101 points.
 
[image]

I monitor the time for the scan to complete with the EXSC field (1=scanning, 0=done)

If 13BMD:m29.VELO is 1.15 (920 steps/s) I see this when I do the scan twice

```
(base) [epics@corvette motor]$ camonitor -tc 13BMD:scan1.EXSC
13BMD:scan1.EXSC               (2024-12-31 11:09:31.621315) 1
13BMD:scan1.EXSC               (2024-12-31 11:09:35.525949) 0
13BMD:scan1.EXSC               (2024-12-31 11:09:36.679780) 1
13BMD:scan1.EXSC               (2024-12-31 11:09:40.585258) 0
```
It takes 3.9 seconds to complete each scan, about 25 points/s.

If 13BMD:m29.VELO is 1.16 (928 steps/s) I see this:
```
13BMD:scan1.EXSC               (2024-12-31 11:09:59.694063) 1
13BMD:scan1.EXSC               (2024-12-31 11:10:47.235558) 0
13BMD:scan1.EXSC               (2024-12-31 11:10:54.149164) 1
13BMD:scan1.EXSC               (2024-12-31 11:11:40.231517) 0
```
It takes 47 seconds to complete the scan, about 2 points/s.

So incrementing VELO from 1.15 to 1.16 increases the scan time by more than a factor of 10.  Very strange, since in single-step mode the VELO field should be irrelevant.

MarkRivers, 2024-12-31:
I have done some more experimenting, enabling asynTrace on the TCP port.  and I have found that the problem is a function of the motor record VELO and ACCL fields, as well as the number of steps to move.

If VELO=1.0 (800 steps/s), ACCL=.174 seconds, and RVAL is incrementing by 1 then there is no delay and the command stream on the TCP port is the following.  Note that the time between PAE commands is 40 ms.
```
2024/12/31 13:10:52.929 10.54.160.149:23 TCP write 8
PAE=368\r
2024/12/31 13:10:52.929 10.54.160.149:23 TCP read 1
:
2024/12/31 13:10:52.929 10.54.160.149:23 TCP write 52
ACE=4096;DCE=4096;SPE=800;limdcE=319488;nrmdcE=4096\r
2024/12/31 13:10:52.929 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:10:52.929 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:10:52.930 10.54.160.149:23 TCP read 1
:
2024/12/31 13:10:52.968 10.54.160.149:23 TCP write 8
PAE=369\r
2024/12/31 13:10:52.969 10.54.160.149:23 TCP read 1
:
2024/12/31 13:10:52.969 10.54.160.149:23 TCP write 52
ACE=4096;DCE=4096;SPE=800;limdcE=319488;nrmdcE=4096\r
2024/12/31 13:10:52.969 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:10:52.969 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:10:52.969 10.54.160.149:23 TCP read 1
:
2024/12/31 13:10:53.007 10.54.160.149:23 TCP write 8
PAE=370\r
2024/12/31 13:10:53.007 10.54.160.149:23 TCP read 1
:
2024/12/31 13:10:53.007 10.54.160.149:23 TCP write 52
ACE=4096;DCE=4096;SPE=800;limdcE=319488;nrmdcE=4096\r
2024/12/31 13:10:53.008 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:10:53.008 10.54.160.149:23 TCP write 4
BGE\r
```
If ACCEL is decreased from .174 to .173 the command stream is the following. Note that the time between PAE commands is 510 ms.  The difference in the commands is that ACE and DCE have increased from 4096 to 5120.
```
2024/12/31 13:15:29.434 10.54.160.149:23 TCP write 8
PAE=368\r
2024/12/31 13:15:29.434 10.54.160.149:23 TCP read 1
:
2024/12/31 13:15:29.435 10.54.160.149:23 TCP write 52
ACE=5120;DCE=5120;SPE=800;limdcE=319488;nrmdcE=5120\r
2024/12/31 13:15:29.435 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:15:29.435 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:15:29.435 10.54.160.149:23 TCP read 1
:
2024/12/31 13:15:29.942 10.54.160.149:23 TCP write 8
PAE=369\r
2024/12/31 13:15:29.942 10.54.160.149:23 TCP read 1
:
2024/12/31 13:15:29.942 10.54.160.149:23 TCP write 52
ACE=5120;DCE=5120;SPE=800;limdcE=319488;nrmdcE=5120\r
2024/12/31 13:15:29.942 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:15:29.942 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:15:29.943 10.54.160.149:23 TCP read 1
:
2024/12/31 13:15:30.450 10.54.160.149:23 TCP write 8
PAE=370\r
2024/12/31 13:15:30.450 10.54.160.149:23 TCP read 1
:
2024/12/31 13:15:30.450 10.54.160.149:23 TCP write 52
[... 80 lines cut ...]
2024/12/31 13:28:03.235 10.54.160.149:23 TCP write 55
ACE=13312;DCE=13312;SPE=800;limdcE=319488;nrmdcE=13312\r
2024/12/31 13:28:03.236 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:28:03.236 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:28:03.236 10.54.160.149:23 TCP read 1
:
2024/12/31 13:28:03.742 10.54.160.149:23 TCP write 8
PAE=282\r
2024/12/31 13:28:03.743 10.54.160.149:23 TCP read 1
:
2024/12/31 13:28:03.743 10.54.160.149:23 TCP write 55
ACE=13312;DCE=13312;SPE=800;limdcE=319488;nrmdcE=13312\r
2024/12/31 13:28:03.744 10.54.160.149:23 TCP read 5
:::::
2024/12/31 13:28:03.744 10.54.160.149:23 TCP write 4
BGE\r
2024/12/31 13:28:03.744 10.54.160.149:23 TCP read 1
:
```

It appears to me that there is a problem in the DMC-4183 firmware when making very short moves. If the computed value of AC and DC are greater than a certain value then the move takes about 500 ms, rather than 40 ms.

@motorapp is this a problem you have seen before?  Do you think I am correct about a firmware issue?  Is there anything we can do in the driver to avoid it?

motorapp, 2025-01-01:
Yes, I've seen this issue before.  Yes, I think it's due to the firmware mode of operation.
The user manual mentions the "motion profiler" within the Galil will generate either a trapezoidal or triangular velocity profile.  Acceleration, velocity, then deceleration.  This suggests the "motion profiler" might work best with 3 steps or more in the move.  Can the microstepping be increased in this case to compensate for the suggestion that 3 steps might be required?  This may also suggest that it works best with an odd number of steps in the step move, excepting 1 step.

The normal, and limit/emergency acceleration, deceleration is being set for every step move at present.  This will also cause some delay, but unrelated to the main issue.

MarkRivers, 2025-01-01:
> Can the microstepping be increased in this case to compensate for the suggestion that 3 steps might be required?

In most cases no, we are using the Galil to generate pulse/direction for existing half-step driver cards.  I don't want to have to replace those cards.

This will be a problem when we want to scan with very fine steps of 1 or 2 motor pulses.  Their profile generator should be smart enough to quickly execute a 1 or 2 step move, without 500 ms delay.

I need to be able to write a little sequence of instructions that will demonstrate the problem to Galil.  The trick is that with the EPICS driver the status information signaling the move is complete is coming over the UDP port, not the TCP port.  Is it possible to disable UDP and have the TCP port do the polling?  It looks like it from the comments in the startup script.  That would be an easier test case to send them.

More generally, is there more documentation on using microcode and other advanced features with the EPICS driver? The documentation mentions .gmc files.  But I see .dmc files in the iocBoot directories.  What is the difference between .gmc and .dmc?

motorapp, 2025-01-02:
Within file iocBoot/iocGalilTest/DMC01Configure.cmd there is this line 
epicsEnvSet("UPDPERIOD", "8")
To force TCP, change the 8 to -8, this will tell the driver to use TCP only.  For positive update period values the driver will try UDP and fall back to TCP, if UDP isn't successful.

The gmc file extension is used by earlier versions of the Galil EPICS driver for Galil micro-code files.
The software written by Galil uses the dmc extension instead, so the driver was changed to match it a while back.
I'm considering removing references to the gmc extension in the driver, but it could impact older installations at the Australian Synchrotron.

I agree with your assessment that this is a Galil profiler error.  I'd be interested in what Galil can come up with (possibly an NRE).  NRE = Non-recurring engineering request.  Establishing a case they can reproduce in their office is essential.  Please let me know how this investigation progresses, I am very interested.

Unfortunately, there is little documentation explaining the more advanced features of the driver.  The startup command scripts and the template/substitution files are a good place to look for comments.  For example, the comments in iocBoot/iocGalilTest/DMC01Configure.cmd explain the IOC shell command parameters.

motorapp, 2025-01-02:
Thinking out load here...
1. One solution could be to ask Galil for a method to request a square velocity profile, without acceleration/deceleration.
Where the motor jumps straight from 0 to the specified velocity, then back to 0.  
2. Or a method of disabling acceleration/deceleration.

motorapp, 2025-01-02:
PVT mode shows some potential to solve this case.  The manual mentions that acceleration, speed settings for independent motion don't apply to PVT mode.  I could use some help with testing, as I don't have access to a 4000 series controller.  Do you have time to help?

Attached is a modified/temporary version of GalilAxis.cpp.  The GalilAxis::move and GalilAxis::beginMotion methods have been altered to use PVT mode for the move.  The Galil PVT command format specifies time in controller samples where the highest allowed value is 2048 and the lowest is 2 samples.  The controller samples every 976us with TM1000 which is the default.  With this in mind, the highest step speed anticipated is < 500 Steps/Second using this approach.  Do you have time to try the code modifications on a test controller?

The modifications to GalilAxis make it unsuitable for normal independent moves due to controller PVT limitation and sample time.  If the method gives improved results for single step scans, we can discuss how PVT could be integrated.  For example, we could put in a record that controls if PVT or normal independent moves are placed by GalilAxis::move.
[GalilAxis.cpp.txt](https://github.com/user-attachments/files/18289298/GalilAxis.cpp.txt)

motorapp, 2025-01-03:
Another approach could be to generate a PVT profile and use that for the scan.  Profiles using linear mode probably won't work, but PVT should be OK.  Software to help generate various profiles would be handy in many cases.

mp49, 2025-01-06:
Hi Mark, do you know if the moves are actually being executed by the amplifier? I'm wondering if there is some problem bit set in any of the status registers, or perhaps some parameter limit is being reached (torque, accel, etc.).

motorapp, 2025-01-30:
I believe issue #35 was solved by Robin Riley and Mark Rivers in Issue #37.
Please reach out if there are any concerns.

closed 2025-01-30 by motorapp