motorapp/Galil #36: Incorrect pulse output when single-stepping stepper motors on DMC-4183

MarkRivers, 2025-01-05:
This problem occurs when using the DMC-4183 with the pulse and direction outputs to run external stepper drivers.  

This is a description of the problem:
- The problem is most evident when moving the motor in single steps with the PAm=N command, and N is incrementing or decrementing by 1 in successive commands.
- The problem only occurs on motors E-H, we have not observed it on motors A-D.
- The controller occasionally (1-10% of the time) puts out 2 or 3 pulses in rapid sequence for the move, rather than just 1.  This results in the motor moving erratically.
- The problem does not occur for all controllers, but does occur for 4 out of 8 controllers that we have tested so far.

I tested this as follows:
- Connected the Pulse Output to both an oscilloscope and to a pulse counter instrument.  That signal also goes to the stepper driver card, via a circuit board that converts the signal from single-ended to differential.
- Connected a test motor to the channel under test on the driver box.  The driver (Step-Pak SPD-6U) is in half-step mode, and this is a 200 step/rev motor, so 400 Galil steps is 1 revolution.
- Used a script to single-step the motor +400 steps and then -400 steps.  The script just increments or decrements the target position by 1 and waits for the motor to be done, doing this 400 times.  Each step takes about 45 ms.
- Observed the number of pulses counted by the pulse counter during each move.
- Observed the test motor to see if it moved 1 full revolution in both the forward and reverse directions.
- Observed the pulse train on the oscilloscope.

These were the results.
- All 8 channels of some controllers work fine, which means the following:
  - The scaler counted exactly 400 pulses during each move.
  - The motor moved exactly 1 revolution (as well as my eye could tell) in each direction.
  - The pulse train on the scope was evenly spaced with about 45 ms between pulses.  There were no rapid sequences of pulses with spacings shorter than this.
- On the problematic controllers channels A-D also work fine.
- However, on these controllers channels E-H display the following behavior.
  - The pulse counter always counts more than 400 pulses, typically between 405 and 500.
  - The scope displayed extra pulses as shown in the photo below.  The trace is 50 ms/division, and most pulses are about 45 ms apart.  However, to the left of the center of the trace there are double pulses, with the second pulse about 5-10 ms after the first.  Sometimes there are triple pulses.  These extra pulses explain why the scaler counts more than 400 counts.
  - The motion of the motor is incorrect,  it is missing steps.  The 400 step motion always results in less than 1 revolution of the motor, in spite of the fact that more than 400 pulses are being output.  This is probably because the extra pulses are too close together, so it does not move correctly.
   - The Galil controller reports that the motor moved exactly 400 steps, even though more than 400 pulses are output.

[image]

This seems like this is most likely a problem in the Galil firmware.  Other possibilities include:
 - It could be an issue in this EPICS driver, but then it would be very strange that axes A-D work and E-H don’t.
- It seems unlikely to be a “bad” unit, because 4 out of 8 controllers tested show the behavior.  
- We do have a circuit on each pulse and direction output that converts the signal from single-ended to differential, but the pulse output I am looking at on the scope is the input to that circuit, i.e.  the single-ended signal from the Galil.  The extra pulses on the scope trace look like nice clean pulses, not noise or reflections.  These are individual circuit boards for each channel, so it would be strange that channels A-D work and E-H do not.

The 3 screenshots below show the EPICS configuration:
- Controller_screen is the configuration of the entire controller.
- Axis_extras_screen is the extra axis configuration parameters.
- Axis_motion_screen is the axis motion configuration.

Perhaps these can give some clue to the issue.  Could the step smooth factor (1.313) have anything to do with this?
[image]
[image]
[image]

motorapp, 2025-01-06:
The controller firmware revision is shown at the top of the galil_dmc_ctrl.adl screen (see below).
The controller behaviour inconsistency could be informative.
Could you check the good vs bad controllers to see if there are firmware differences?  Thanks.

The controller firmware shown for the bad controller is 1.3K.  This is the most recent firmware for 41x3. 
It is possible to upgrade/downgrade the firmware.
Here is a link to the Galil firmware download page (some revision history):
https://www.galil.com/downloads/firmware

The KS step smoothing constant is shown on the motor extras screen, about 1/2 way down.
KS recommended value is 0.5 (different from driver default)
Things to check.
MG _TM - Time base.  TM1000 is default

Other commands worth checking are:
IT - SCURVE recommend value 1 (no filtering)

There is a command console at the bottom of the galil_dmc_ctrl.adl screen.  Galil commands can be issued through this console.

Let me know what you find.  Thanks.)
[image]

MarkRivers, 2025-01-06:
All controllers are running the same firmware, 1.3k.

motorapp, 2025-01-06:
Lets try with the KS and IT commands.
[image]

The IT command isn't supported by the driver.  Please use the command console on the controller screen or use GalilTools to set it.  Also, burn the updated value to eeprom with BS command.

motorapp, 2025-01-06:
The step pack will require a minimum high step pulse width.  It will be related to the maximum step input frequency allowed on the drive.
The Galil should provide a 50% duty cycle step pulse output.  If the step rate is too high (SP), the high pulse width could become too narrow for the step pack.  It could be the result of KS, IT or SP settings.

motorapp, 2025-01-07:
Galil Design Kit software can be used to upload all the controller settings.  This upload could be performed on both the good and bad controller, then a diff tool could spot any differences in settings.

motorapp, 2025-01-07:
Does the installation use SSI or BISS encoders at all?

MarkRivers, 2025-01-08:
Galil is setting up to reproduce the problem.

I think that a DMC-4183 may be composed of 2 cards: a 4-channel main card with Ethernet, USB, etc. and a 4-channel expansion card. Perhaps the issue is the connection between those cards? That would explain why the problem is always on the last 4 channels, and why some controllers are fine.

motorapp, 2025-01-08:
For the purpose of approaching Galil about this issue.  Recommend aligning the software approach with Galil as much as possible.  Eliminate as much software as possible from the stack, excluding Galil software.

For example, do the motion control through Galil commands alone.  Gather data through any means required.

MarkRivers, 2025-01-08:
I have done that. I captured the complete TCP communication with the Galil and sent it to them with the trace data.  It is basically 400 repetitions of

PAH=1
BGH
PAH=2
BGH

Robin understands that this is not a software problem, it is a hardware or firmware problem.

motorapp, 2025-01-09:
It's great that Robin (Galil) is investigating.  I hope a solution is found quickly.
As you know, the controllers may have to be shipped back to California.
Keep us updated, if you have time.

MarkRivers, 2025-01-10:
I have tested 40 of our 49 controllers so far.  19 have the problem, 21 are OK.  

The problem is not restricted to the step and direction outputs.  8 of the controllers I tested are using the on-board 4141 or 4040 stepper drivers.  4 of those controllers also have the problem.  I tested by single-stepping a test motor for 1 full revolution in each direction.  For the bad controllers it works fine on channel A, but fails on channel H, where it typically only completes 75-90% of a revolution.

MarkRivers, 2025-01-11:
All 49 controllers now tests, 23 bad, 26 OK.

motorapp, 2025-01-13:
One would hope for better odds than this.  Must be a hassle for the team.  I hope user beam time isn't being interrupted.  Let's hope for some promising feedback from Galil soon.

MarkRivers, 2025-01-15:
I had previously tested that changing KS to larger values did not make a difference.  Today I did a more systematic test on a problematic channel.  I did 400 single-steps in the positive direction while capturing the step and direction lines on the USB-CTR08 a 1 kHz.  I found that KS=1,2,4,8 still had the problem.  However, KS=16 fixed the problem.

I am very happy to have found a workaround.  I have changed KS to 16 on all 92 problematic channels.  It seems to fix the problem on all motors I have tested.

Galil still needs to determine why channels E-H on 50% of our controllers require this setting, while the others work fine with KS=1.313.

@motorapp can you explain why the default value of STEPSSMOOTH_SP is 1.313?  Robin Riley recommended values that are powers of 2, e.g. 1, 2, 4, 8, 16, 32, 64.  STEPSMOOTH_SP also has DRVH=16, but Robin suggested using 32 or 64.

motorapp, 2025-01-16:
It's good news a workaround has been found.  Let's hope the workaround holds for all the required circumstances.
The default value of 1.313 for the step smooth value comes from the older 2183 command reference manual.  The 2183 command reference manual specifies the default of 1.313.  As you know, the database default of 1.313 only applies constantly if autosave isn't used.  The step smooth factor is saved/restored by autosave.  The relevant request files are:
GalilSup/Db/galil_motor_extras.req
iocBoot/iocGalilTest/GalilTest_DMC01Settings.req

The STEPSMOOTH_SP DRVH has been changed to 128 to accommodate the largest value supported by surveyed controllers.
The STEPSMOOTH_SP DRVL has been changed to to 0.25 to accommodate the smallest value supported by surveyed controllers.
I checked 21x3, 41x3, 40x0, 500x0 and 30000 series controllers.

motorapp, 2025-01-30:
I believe issue #36 was solved by Robin Riley and Mark Rivers in Issue #37.
Please reach out if there are any concerns.

closed 2025-01-30 by motorapp

MarkRivers, 2025-02-12:
> I believe issue https://github.com/motorapp/Galil/issues/36 was solved by Robin Riley and Mark Rivers in Issue https://github.com/motorapp/Galil/issues/37.

This issue is not currently solved, but it is a hardware problem, not an issue with the Galil EPICS driver.

Galil has finally (after 4 weeks) reproduced the problem of incorrect step and direction output signals on channels E-H.  Their hardware group is currently investigating.  Based on statistics from me and other customers at the APS it appears that about half of the DMC-4183 units in the field do not correctly drive stepper  motors on channels E-H.