motorapp/Galil #37: Stepper motors continue to move after DMOV=1

MarkRivers, 2025-01-16:
When using a stepper motor with the DMOV signal is set to 1 before the controller is done outputting step pulses.  This problem gets larger as KS increases.

The reason for this is explained in the Galil DMC-41x3 manual starting on page 91.

> For proper controller operation, it is necessary to make sure that the controller has completed generating all step
pulses before making additional moves. This is most important when back and forth motion is commanded. For
example, when operating with servo motors, the trippoint AM (After Motion) is used to determine when the
motion profiler is complete and is prepared to execute a new motion command. However when operating in
stepper mode, the controller may still be generating step pulses when the motion profiler is complete. This is
caused by the stepper motor smoothing filter, KS. To understand this, consider the steps the controller executes to
generate step pulses:

> When used in stepper mode, the MC command will hold up execution of the proceeding commands until the
controller has generated the same number of steps out of the step count register as specified in the commanded
position. The MC trippoint (Motion Complete) is generally more useful than AM trippoint (After Motion) since the
step pulses can be delayed from the commanded position due to stepper motor smoothing.

I have found that I need to use KS=16 to eliminate problems on some channels.  I also found that even for "normal" channels KS=16 allows running motors faster without stalling.  For example, one of our beamline motors stalled at about 6,000 steps/s with KS=1.313 but was able to run over 12,000 steps/s with KS=16.

The problem is that with KS=16 the EPICS driver is setting DMOV=1 long before the controller is done outputting step pulses.  I tested this as follows:
- Connect step signal to an EPICS binary input that is being sampled at 500 Hz with SCAN=I/O Intr.
- Run camonitor on the motor DMOV signal and the bi record with the step signal.
- Test a motor with MRES=1, ACCL=0.5, VELO=2000.
- Observe what happens around the time when the DMOV changes back to 1 (motion complete).
- Motor stop delay (STOPDELAY_SP) = 0.
- Move 1000 steps positive.

This is with KS=1.  It is OK, there are not steps after DMOV=1.
```
13MCBOX:USBCTR:Bi1             (2025-01-16 16:02:49.463768) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:02:49.465805) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:02:49.468939) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:02:49.474937) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:02:49.476943) Low
13LAB:m1.DMOV                  (2025-01-16 16:02:49.481836) 1
```
This is KS=2.  There is 1 extra pulse about 8 ms after DMOV=1.
```
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.135178) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.139258) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.140468) High
13LAB:m1.DMOV                  (2025-01-16 16:04:04.140608) 1
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.144899) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.148865) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:04:04.152900) Low
```
This is with KS=4.    There are 2 extra pulses, the last about 25 ms after DMOV=1.
```
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.361225) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.364953) High
13LAB:m1.DMOV                  (2025-01-16 16:06:11.365514) 1
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.369215) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.371248) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.378007) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.385355) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:06:11.389582) Low
```
This is with KS=8.    There are 7 extra pulses, the last about 70 ms after DMOV=1.
```
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.000963) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.002864) High
13LAB:m1.DMOV                  (2025-01-16 16:08:13.003550) 1
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.004811) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.007488) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.009667) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.011635) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.014716) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.018975) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.020788) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:08:13.023987) High
[... 53 lines cut ...]
13MCBOX:USBCTR:Bi1             (2025-01-16 16:13:47.511922) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:13:47.524000) Low
13MCBOX:USBCTR:Bi1             (2025-01-16 16:13:47.536648) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:13:47.539766) Low
13LAB:m1.DMOV                  (2025-01-16 16:13:47.542516) 1
```
Now DMOV goes to 1 about 3 ms after the last step is complete.  

But now if I just take 1 step I see the following.  It is waiting 125 ms after the last step, which is a waste of time, and will slow down scans with small step sizes.
```
13LAB:m1.DMOV                  (2025-01-16 16:15:24.404584) 0
13MCBOX:USBCTR:Bi1             (2025-01-16 16:15:24.457577) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:15:24.462532) Low
13LAB:m1.DMOV                  (2025-01-16 16:15:24.587417) 1
```
That same 1 step move with STEPDELAY_SP=.02 shows this:
```
13LAB:m1.DMOV                  (2025-01-16 16:18:21.586070) 0
13MCBOX:USBCTR:Bi1             (2025-01-16 16:18:21.639217) High
13MCBOX:USBCTR:Bi1             (2025-01-16 16:18:21.643263) Low
13LAB:m1.DMOV                  (2025-01-16 16:18:21.663013) 1
```
So I could have used a much shorter STEPDELAY_SP for a short move.

It seems like the best solution would be to monitor MC.  Would it be possible to change the driver to do that?

motorapp, 2025-01-17:
Unfortunately, it is not possible.(  Within the MC command reference, at the top is a legend describing when MC command can be applied.  The legend states the MC command is not available from the terminal, only from within Galil micro-code program.  The 41x3 micro-code execution speed makes micro code not attractive as a solution.  The controller response to terminal queries is much faster with 41x3.

[image]

KS may fix some issues, but it is causing other issues.

The driver obtains the axis moving status from the data record sent by the controller.  The moving status within the record is effectively the same as MG _BGx.  This is a non-blocking form of the AM command.  Something similar is required for the MC command, but doesn't exist right now.

Tried the following approach to step scans (recommend KS at default):
1.  caput to PV DMC01:SEND_STR_CMD "IPA=1"    //Assuming motor A and 1 step
2.  Monitor DMOV and wait till 1
3.  Trigger detector, complete acquisition
4.  Return to step 1
Tried a 100 point scan, it took around 11-12 seconds to complete

motorapp, 2025-01-17:
The best approach to scans on Galil is servo motor fly scans using output compare.  Not very useful in this case though.

MarkRivers, 2025-01-24:
Robin Riley from Galil suggested a solution, which uses _TDx and _RPx after the profile is complete. _TXs is the auxilliary encoder which is output step counter for stepper motors.  _RPx is the reference (commanded) position.   _TDx continues to update as pulses are output after the profile is complete.  Only when _TDx = _RPx is the move really complete.

I just tested this using this change to GalilAxis.cpp.

```
diff --git a/GalilSup/src/GalilAxis.cpp b/GalilSup/src/GalilAxis.cpp
index 17cea4e..f5195e0 100644
--- a/GalilSup/src/GalilAxis.cpp
+++ b/GalilSup/src/GalilAxis.cpp
@@ -1666,6 +1666,9 @@ asynStatus GalilAxis::getStatus(void)
    int brakeport;                              //Brake port for this axis
    int limitDisable = 0;                        //Limit disabled param
    bool ctrlType;                              //Controller type
+   int motortype = 0;  // Motor type
+   bool stepper = false;  // True if stepper
+   double reference_position;

    //If data record query success in GalilController::acquireDataRecord
    if (pC_->recstatus_ == asynSuccess) {
@@ -1677,6 +1680,25 @@ asynStatus GalilAxis::getStatus(void)
          strcpy(src, "_BGx");
          src[3] = axisName_;
          inmotion_ = (bool)(pC_->sourceValue(pC_->recdata_, src) == 1) ? 1 : 0;
+         //If this is a stepper the profile might be done, but the pulse output not done.
+         //Check of the step out counter is equal to the target position
+         //Figure out if we have a stepper
+         pC_->getIntegerParam(axisNo_, pC_->GalilMotorType_, &motortype);
+         stepper = ((motortype >= 2) && (motortype <= 5));
+         if (!inmotion_ && stepper) {
+            strcpy(src, "_TDx");
+            src[3] = axisName_;
+            motor_position_ = pC_->sourceValue(pC_->recdata_, src);
+            strcpy(src, "_RPx");
+            src[3] = axisName_;
+            reference_position = pC_->sourceValue(pC_->recdata_, src);
+            if (motor_position_ != reference_position) {
+               asynPrint(pC_->pasynUserSelf, ASYN_TRACE_ERROR,
+                  "GalilAxis::getStatus reference_position=%f motor_position=%f\n", reference_position, motor_position_);
+               inmotion_ = true;
+            }
+         }
+
```
If the motor is not in motion (profiler done) and it is a stepper then it enters the code block that reads RPx and TDx .  If they are not equal it sets inmotion_ to true.

I have tested this with different values of KS. I am polling at 50 Hz (20 ms period). 
- KS=1  Code block almost never entered because the profile and the step output are both complete within one poll cycle.
- KS=2  Code block is almost always entered once and we see this:
```
2025/01/23 18:35:57.452 GalilAxis::getStatus reference_position=16800.000000 motor_position=16799.000000
```
KS=4  Code block is usually entered twice and we see this:
```
2025/01/23 18:37:45.612 GalilAxis::getStatus reference_position=18400.000000 motor_position=18396.000000
2025/01/23 18:37:45.631 GalilAxis::getStatus reference_position=18400.000000 motor_position=18399.000000
```
- KS=8 code block is usually entered 3 times and we see this:
```
 2025/01/23 18:38:34.916 GalilAxis::getStatus reference_position=19200.000000 motor_position=19193.000000
2025/01/23 18:38:34.935 GalilAxis::getStatus reference_position=19200.000000 motor_position=19197.000000
2025/01/23 18:38:34.955 GalilAxis::getStatus reference_position=19200.000000 motor_position=19199.000000
```
- KS=16 code block is entered 8 times and we see this
```
2025/01/23 18:39:41.546 GalilAxis::getStatus reference_position=20000.000000 motor_position=19978.000000
2025/01/23 18:39:41.566 GalilAxis::getStatus reference_position=20000.000000 motor_position=19984.000000
2025/01/23 18:39:41.585 GalilAxis::getStatus reference_position=20000.000000 motor_position=19989.000000
2025/01/23 18:39:41.605 GalilAxis::getStatus reference_position=20000.000000 motor_position=19993.000000
2025/01/23 18:39:41.624 GalilAxis::getStatus reference_position=20000.000000 motor_position=19995.000000
2025/01/23 18:39:41.644 GalilAxis::getStatus reference_position=20000.000000 motor_position=19997.000000
2025/01/23 18:39:41.663 GalilAxis::getStatus reference_position=20000.000000 motor_position=19998.000000
2025/01/23 18:39:41.683 GalilAxis::getStatus reference_position=20000.000000 motor_position=19999.000000
```
Note that the motion was not really complete until about 140 ms after _BGX said the move was complete.

This modified code fixes the problem, DMOV will not be set to 1 until the steps have been output.

The code needs to be checked to see if it works when the motor stops for other reasons:
- Limit
- Manual stop command.

It also needs to be checked to see if it works with steppers with encoders.  I only tested with no encoders.

motorapp, 2025-01-24:
Great solution.  Thanks to Robin and yourself for revealing it.  Nice work.  Grateful to you both.

During stop, it seems the reference position and motor position are set equal at move completion by the controller.  This should hold true for manual, limit, and CSAxis emergency stop (eg. 1 motor hits limit, all are stopped).

With encoded steppers, the controller runs them in open loop.  At move completion (inmotion_=false), the driver will synchronize the step count register with the encoder value.  The reference position is set at the same time.  Afterward done and finally dmov are set.

GalilAxis had a member variable ctrlUseMain_.  I've renamed it to be motorIsServo_ for clarity, and used it to implement the prescribed solution.  Testing produced positive results.

I've added the appropriate change log at the top of GalilController.cpp.

Thanks again to Robin and yourself for revealing this solution.))

motorapp, 2025-01-30:
Resolved by Robin Riley and Mark Rivers

closed 2025-01-30 by motorapp