motorapp/Galil #38: Problem with DMC-4080 and V4-0-15

MarkRivers, 2025-02-14:
We are trying to run a DMC-4080 with V4-0-15.  It mostly works.  However, the motor STAT field shows COMM error.  It takes several tries to make a motor move, setting SPMG=STOP and then GO, and then moving the motor.  The MSTA "moving" bit does not change state when the motor is moving.

Were there some changes we made in V4-0-15 when testing the DMC-4183, for example testing that a stepper is really done, that could be broken on the DMC-4080?

motorapp, 2025-02-14:
After reviewing the code changes for V4-0-15 I've been unable to identify an issue.
Recommend the following steps.
1.  Check the 4080 controller firmware version.  If existing firmware <=1.2i, I recommend changing to 1.2H firmware.
     GalilTools or Galil Design Kit software can be used to change the controller firmware.  If anything fails, look at the upgrade jumper.
     The firmware can be downloaded from the weblink below:
     https://www.galil.com/downloads/firmware
2.  Enable communications logging.  Uncomment line 8 of file iocBoot/iocGalilTest/DMC01Configure.cmd.  Inspect 
     iocBoot/iocGalilTest/galil_debug.txt for errors.
3.  Monitor PV $(P)ERROR_MON, inspect for errors.
4.  Check network connectivity, cables.

If the above doesn't help.
1.  Is the network busy?  If yes, check out the network configuration guide.

Recommend UDP 8-20ms update on 4080.  TCP as backup attempt (-8ms).

Good luck, and let me know how it progresses.  Thanks. )

MarkRivers, 2025-02-14:
@motorapp thanks for the reply.  The DMC-4080 is running firmware 1.3i-SER.

This is what we see on the motorx_all screen.  Note that it is showing Comm. Failure.
[image]

This is the msta screen.  Note that it does NOT show the COMM status bit set.
[image]

Why does STAT=COMM if the COMM bit in MSTA is not set?

When we move the motor the DMOV and MOVING in MSTA do not change state.

If we run Galil 3-6 (provided by the vendor who OEMed the DMC-4080 we don't have the problem with MSTA and STAT, but we do have other problems.  The main controller screen shows controller error, and it does not show the controller model or firmware version.

MarkRivers, 2025-02-14:
Problem solved.  Galil/configure/RELEASE.local was pointing to Galil-3-6 but we were working in Galil-4-0-15.  It was mixing things from 2 releases.

Sorry for the bother.

closed 2025-02-14 by MarkRivers

motorapp, 2025-02-15:
Great to hear it's working now. :)