motorapp/Galil #39: Driver crashes when initializing a brushless servo motor with the BZ command

MarkRivers, 2025-03-25:
I am trying to use a brushless servo motor with the 43547 amplifier.  To do that it needs to be initialized.  I am doing that with interactive commands in the galil_dmc_ctrl.adl screen.

I type the following commands OK:
```
BAA
BMA=1820444.44
BZ <1000>1500
```
Those commands configure axis A as a brushless motor, sets the encoder counts per magnetic cycle, and sets the time for the 2 steps of the initialization.

I then type this command which initializes the axis with 3V control voltage:
```
BZA=3
```
When I type that command the motor moves twice, with a few degrees of motion.

However, the Galil driver then crashes:
```
Disconnected from DMC4183 Rev 1.3k at 10.54.160.215
Connected to DMC4183 Rev 1.3k at 10.54.160.215
Segmentation fault (core dumped)
```
I suspect that perhaps the Galil does not respond to poll commands when it is executing the BZA command, and perhaps that is what is crashing the driver?

Has the driver been used with brushless motors being initialized like this before?

motorapp, 2025-03-26:
Responding to the questions.
A)  Yes, execution of the BZ command will disrupt communication with the driver.  From the output provided, I suspect it's crashing somewhere in GalilStartController.  After reconnecting with the controller, GalilStartController is executed, and I can't see it's output in the provided list.  The GalilStartController expected output is "Code started successfully on model....".  Another method is recommended for initializing brushless motors, refer B below.
B)  Yes, the driver is used with brushless motors on many sites.  The recommended commutation initialization method is to use Galil micro-code.
The driver maintains a variable in the micro-code cinit<Axis>, that signifies if the axis commutation has been initialized successfully or not.
Recommend creating a init.dmc file with the following contents.  

BAA
BMA=1820444.44
BZ <1000>1500
IF(cinitA=0)
BZA=-3
ENDIF
IF(_MOA);cinitA=1;ELSE;cinitA=0;ENDIF

Note the BZA=-3 change in the above code.  The -3 means that if BZ executes successfully, then the motor amplifier will be enabled afterward.  If BZ executes unsuccessfully, then the motor amplifier won't be enabled afterward.  The code checks to see if the motor amplifier is enabled after BZ, if it's enabled, then cinitA is set 1 to indicate successful commutation initialization.

Add the custom code above to the generated code by using GalilAddCode command in the ioc start script (eg. iocBoot/iocGalilTest/DMC01Configure.cmd).  
For this case, GalilAddCode should appear after GalilCreateController, but before any GalilCreateAxis calls.  The GalilAddCode call should appear as below:
GalilAddCode("$(PORT)", 0, "init.dmc")

Second parameter = code section for custom code = 0 card code = code is inserted at line 2, after the label #AUTO.  The custom code is executed only upon micro-code start, it is not executed during operations.

The generated + custom code can be viewed in iocBoot/iocGalilTest/<controllerip>_prd.dmc file, or by viewing the code downloaded to the controller.

Let me know how it progresses, thanks.!

motorapp, 2025-03-26:
For brushless axis A, I recommend using the "Off On Error" function.
The "Off On Error" option is on the motor extras screen, about 1/2 way down.  
Recommend setting "Off On Error" to "Position Error".
The allowed position error is specified in encoder counts, and is found on the motor extras screen, 1 down from the "Off On Error" option.

motorapp, 2025-03-27:
The 4080 command reference manual version 1840, under the BZ command has the remark below.

"While the BZ command is executing, DMC code, data records, and communication from the controller will pause until completion"

The poller in the Galil driver will disconnect after 3 consecutive errors when obtaining the data record.  The timeout period is 1 second.  I'd imagine issuing the BZ command through the DMC controller screen command console, would be similar to removing the network cable, then reconnecting it.  The driver is designed (excluding bugs) to handle this.

MarkRivers, 2025-03-27:
>  I'm also not sure if UDP mode was used there when the crash was encountered. I've assumed UDP was used given the nature of the code fault discovered.

Yes, I was using UDP mode when it crashed.

> Recommend the micro-code solution for initializing the brushless axis. 

I do plan to use the micro-code solution in the long run.  However, for initial testing I want to do it interactively to make sure I am using the correct parameters.

I have downloaded your fix and I'll let you know if it still crashes.  The motor is not currently available for testing, but I should have another one soon.

motorapp, 2025-03-28:
> Yes, I was using UDP mode when it crashed.

Thanks for the confirmation.

> I do plan to use the micro-code solution in the long run.  However, for initial testing I want to do it interactively to make sure I am using the correct parameters.

For sure.  Also, if the command console PV is provided by the driver, it should work with most commands ideally.

> I have downloaded your fix and I'll let you know if it still crashes.  The motor is not currently available for testing, but I should have another one soon.

Grateful for your help.  Thank you.

motorapp, 2025-05-14:
Hi Mark,
I hope all is well?  Are we okay to close this issue or is more time required?  Thanks!
Kind regards,
Mark Clift

motorapp, 2025-09-26:
I've tried replacing the atof command in the command console code with strtod.  The change is in the branch ELO-AZ1_recovery_issue.

The part about the driver crashes when BZ is issued from a telnet session is interesting.
Does removing the Ethernet cable cause the driver to crash in the same way?