motorapp/Galil #42: Issue AZ1 command to recover from ELO signal

MarkRivers, 2025-09-22:
I want to implement a hardware emergency stop using the ELO (electronic lock-out) signal to the DMC-4183 on the 44-pin connector.  That is working.  However, to recover from ELO latched state one must do the following:
- De-assert the ELO signal
- Disable all axes
- Send the AZ1 command.
- Re-enable all axes

It looks like the driver does not provide a way to send the AZ1 command except from the command line.  It would be nice to add that.

Also, the driver appears to decode all of the amplifier error status bits:
```
        //Amplifier error status
        map["TA00"] = Source(52, "UB", 0, "Boolean", "Axis A-D over current");
        map["TA01"] = Source(52, "UB", 1, "Boolean", "Axis A-D over voltage");
        map["TA02"] = Source(52, "UB", 2, "Boolean", "Axis A-D over temperature");
        map["TA03"] = Source(52, "UB", 3, "Boolean", "Axis A-D under voltage");
        map["TA04"] = Source(52, "UB", 4, "Boolean", "Axis E-H over current");
        map["TA05"] = Source(52, "UB", 5, "Boolean", "Axis E-H over voltage");
        map["TA06"] = Source(52, "UB", 6, "Boolean", "Axis E-H over temperature");
        map["TA07"] = Source(52, "UB", 7, "Boolean", "Axis E-H under voltage");

        map["TA1A"] = Source(53, "UB", 0, "Boolean", "Axis A hall error");
        map["TA1B"] = Source(53, "UB", 1, "Boolean", "Axis B hall error");
        map["TA1C"] = Source(53, "UB", 2, "Boolean", "Axis C hall error");
        map["TA1D"] = Source(53, "UB", 3, "Boolean", "Axis D hall error");
        map["TA1E"] = Source(53, "UB", 4, "Boolean", "Axis E hall error");
        map["TA1F"] = Source(53, "UB", 5, "Boolean", "Axis F hall error");
        map["TA1G"] = Source(53, "UB", 6, "Boolean", "Axis G hall error");
        map["TA1H"] = Source(53, "UB", 7, "Boolean", "Axis H hall error");

        map["TA2A"] = Source(54, "UB", 0, "Boolean", "Axis A at _TKA peak current");
        map["TA2B"] = Source(54, "UB", 1, "Boolean", "Axis B at _TKB peak current");
        map["TA2C"] = Source(54, "UB", 2, "Boolean", "Axis C at _TVC peak current");
        map["TA2D"] = Source(54, "UB", 3, "Boolean", "Axis D at _TKD peak current");
        map["TA2E"] = Source(54, "UB", 4, "Boolean", "Axis E at _TKE peak current");
        map["TA2F"] = Source(54, "UB", 5, "Boolean", "Axis F at _TKF peak current");
        map["TA2G"] = Source(54, "UB", 6, "Boolean", "Axis G at _TKG peak current");
        map["TA2H"] = Source(54, "UB", 7, "Boolean", "Axis H at _TKH peak current");

        map["TA3AD"] = Source(55, "UB", 0, "Boolean", "Axis A-D ELO active");
        map["TA3EH"] = Source(55, "UB", 1, "Boolean", "Axis E-H ELO active");
````

But these are not exposed as EPICS PVs, correct?  It would be nice to add those, and an medm screen to display them.

motorapp, 2025-09-24:
Hi Mark,
The PV's have been added, and I've updated the amplifier screen.

I'm still working through some GUI issues on MEDM and QT, related to organization and colors.
If you have a chance in the meantime, could you let me know if this approach meets the requirement?
Kind regards,
Mark Clift

[image]

motorapp, 2025-09-26:
Hi Mark,
The branch attached to this issue includes the features requested in this issue.  As time allows could you give it a try?

Other changes are also included in the attached branch.
1.  If a motor amplifier faults during movement, the axis will be stopped using limit deceleration.  If the axis is part of a CSAxis, the group will be stopped using limit (emergency) deceleration.  The internal stop mechanism has been updated to include this case and others found in the Galil command SC reference manual.  The limit deceleration is defined in motor extras by the field "EGU after limit hit".
2.  A major change in the branch is display colors, as shown below.
Fault status monitors are now green when false, red when true (alarm status colors now)
Impacts EtherCat, BISS, and amplifier status medm screens
Additional color changes that impact Qt only are:
Limit status -> treated as fault status (now red when active) (MSTA bits are not treated as fault status)
Moving status -> treated as fault status (now red when moving)

Monitor status bits that are not treated as fault status are green when true, red when false (eg. MSTA).

Other changes are bug fixes or enhancements that should (in theory) be transparent.

Let me know if changes are required.
Kind regards,
Mark Clift.

MarkRivers, 2025-09-26:
I tested this branch.  It does not seem to be working correctly.  I have a switch connected to the ELO input on the A-D 44 pin connector.  When the switch is Low and I restart the controller and the IOC I see this.  All the status lights are green, which is correct.
[image]

I then flip the switch to High and I see this.  I expect to see ELO red, perhaps on both A-D and E-H because they both disable all channels, so they may be wired together internally.  However, I actually see OverCurrent and OverVolt on A-D, and OverTemp on E-H.  This seems strange.
[image]

I then flip the switch back to Low and I see this.  OverVolt is back to green, but OverCurrent and OverTemp are still red.
[image]

Pressing "ClearFaults" clears the faults if all motors are disabled.  If one types the AZ1 command manually and the motors are not all disabled then one sees a helpful error message telling you that the motors need to be disabled.  But pressing ClearFaults does not show this helpful message if motors are still enabled.

MarkRivers, 2025-09-26:
Note also that AtTorqueLimit is set for all 8 channels above.  This seems incorrect, since only 2 channels are actually connected to a motor, and they have small torque.

When I use the Galil GDK the status is correct.  This is the view when the ELO input on A-D is High.  It shows ELO active on both AD- and E-H.
[image]

I then flip the ELO switch to Low.  ELO active stays high on A-D, but is cleared on E-H.  I believe this is because there are 2 channels with amplifiers enabled on A-D, but none on E-H.  
[image]

I then type these commands in GDK:
```
MOA  Turn off amplifier A
MOB   Turn off amplifier B
AZ1    Clear ELO error
```
The display then shows this, the ELO on A-D is now cleared
[image]

It seems like the mapping of the record data is somehow wrong (though the code looks correct to me).  I think OverCurrent A-D is actually ELO A-D, and OverCurrent A-D is actually ELO A-H, because it clears when I flip the switch back to Low.  Note also that GDK does not show Peak Current on any channel, while EPICS is showing AtTorqueLimit on all channels.

motorapp, 2025-09-27:
Nice work.)  Update pushed to AZ1 branch for any further integration testing.